thelastoutpostworkshop/gpio_viewer #4: Very cool, but I cannot compile / upload

en-x, 2023-12-30:
This is very cool and I can see where it would be very useful.

Unfortunately, I cannot get it to compile and upload to any of my boards.

The error message I am getting is:

---------------------------------------------------------
In file included from /gpioviewer/gpioviewer.ino:8:
/Arduino/libraries/GPIOViewer/src/gpio_viewer.h: In member function 'int GPIOViewer::readGPIO(int, uint32_t*, pinTypes*)':
/Arduino/libraries/GPIOViewer/src/gpio_viewer.h:297:22: error: 'GPIO' was not declared in this scope
  297 |             value = (GPIO.in >> gpioNum) & 0x1;
      |                      ^~~~
/Arduino/libraries/GPIOViewer/src/gpio_viewer.h:302:22: error: 'GPIO' was not declared in this scope
  302 |             value = (GPIO.in1.val >> (gpioNum - 32)) & 0x1;
      |                      ^~~~

exit status 1

Compilation error: exit status 1
---------------------------------------------------------

I looked at the gpio_viewer.h and it looks like everything is declared property, but not sure if I am doing something wrong of if there is another issue.

Thank you for creating this project. If you have any suggestion on how to fix this issue, I would appreciate your insight.

thelastoutpostworkshop, 2023-12-30:
what is the board your are using ?

en-x, 2023-12-30:
I tried 2 versions of the ESP32-WROOM-32D board (HiLetgo and a DevkitC_v4) and also tried a XIAO ESP32-C3.
All boards give me the same message. I am going to try a different PC with a fresh install of Arduino IDE.

thelastoutpostworkshop, 2023-12-30:
Are you using the Arduino IDE version 2 ?

en-x, 2023-12-30:
Yes. 2.2.1 on a MAC M1.

en-x, 2023-12-30:
I did a fresh install of Arduino IDE 2.2.1 on a windows PC. I added your library .zip file and added the ESPAsyncWebServer library .zip from your link. Then I added the AsyncTCP library by dvarrel version 1.1.4 from the Arduino library manager and you instructed.  This was a completely new install on a PC that never had Arduino IDE on it. and these are the only 3 libraries installed.  The compile failed with the exact same error and I selected the board type that you used in your video tutorial.

Since others seem to have this working, I am assuming I am doing something wrong or missing something.

Anyway, I appreciate you creating this project. It looks like it would have been a useful tool.

Happy New Year.

thelastoutpostworkshop, 2023-12-30:
Happy new Year too!  Last thing I want you to try : compile the example that comes with the library

en-x, 2023-12-30:
Yes, that's what I've been trying. I didn't even change the host or password. Just opened the example and compiled and got the error message. I must be missing a library.

Where is GPIO defined?

thelastoutpostworkshop, 2023-12-31:
it's define in the core library of ESP32

thelastoutpostworkshop, 2023-12-31:
It is is in the header file gpio_struct.h which is in the core library of the ESP32..  The version of this library is 2.0.11.  It is located here \AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\tools\sdk\esp32\include\soc\esp32\include\soc

en-x, 2023-12-31:
Wow!  Thank you so much.  Issue has been resolved!!!

I followed the path on my system and found that I did not have that file in my ESP32 folder. For some unknown reason under board manager: esp32 by Espressif, I had 3.0.0-alpha3 installed and this **does not** include the gpio_struct.h file.  I saw in your path, you are using version 2.0.11, so I went ahead and downgraded my version to 2.0.14 and tried to compile your example code and it worked perfectly.

I really appreciate your help resolving this issue. You've made my day.

Thank you and have a great new year.

thelastoutpostworkshop, 2023-12-31:
Excellent, glad it work now!

closed 2023-12-31 by thelastoutpostworkshop

crittercam, 2024-02-20:
I ran into the same problem.  Your fix resolved it, thanks!