thelastoutpostworkshop/gpio_viewer #7: ESP32 WROOM-32 board support

jlit, 2023-12-30:
Love the library.  Would love support for the [ESP32 WROOM-32D](https://www.amazon.com/gp/product/B08D5ZD528/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1) boards.  Similar looking to the ESP32-VROOM-32D but with different pinout.   

[image]

[image]

[image]

Thanks.

jlit, 2023-12-30:
Here's an image that lines up with the ESP32-VROOM-32D image:
[image]

thelastoutpostworkshop, 2023-12-30:
The board is now available.

closed 2023-12-30 by thelastoutpostworkshop