thelastoutpostworkshop/gpio_viewer #9: esp32s3 in panic / reboot loop

lscalpati, 2023-12-30:
I tried this lib with existing code that already has wifi enabled in the code.  I added the 1st 2 lines of my sketch:

```
#include <gpio_viewer.h>
GPIOViewer gpio_viewer;
```

and added as the last line of my setup():
```
  //init gpioviewer
  gpio_viewer.begin();
```

Here is the serial output with the errors:
```

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4203be3a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
Test V1.1

assert failed: tcpip_api_call IDF/components/lwip/lwip/src/api/tcpip.c:497 (Invalid mbox)

Backtrace: 0x40377e1a:0x3fceb9c0 0x4037b70d:0x3fceb9e0 0x403817f5:0x3fceba00 0x4201ae39:0x3fcebb30 0x4200a13d:0x3fcebb60 0x42009592:0x3fcebbb0 0x42002bdb:0x3fcebbd0 0x4200340e:0x3fcebc30 0x4200ef9a:0x3fcebc50

ELF file SHA256: fb9bc28191471c18

```

thelastoutpostworkshop, 2023-12-30:
Are you using platformIO ?

lscalpati, 2023-12-30:
No using Arduino 1.8.19
I notice that with your lib installed I do get an error at compile time: Error compiling for board ESP32S3 Dev Module
but it continues it upload.  Same exact compile and runtime errors when I use your sample code as the sketch

lscalpati, 2023-12-30:
here is the board info:
```
BN: ESP32S3 Dev Module
VID: 303a
PID: 1001
SN: DC:DA:0C:45:0C:0C

```

thelastoutpostworkshop, 2023-12-30:
Maybe it is related to this : https://github.com/thelastoutpostworkshop/gpio_viewer/pull/5

lscalpati, 2023-12-30:
I am not using PlatformIO and I followed the Arduino instructions exactly

thelastoutpostworkshop, 2023-12-30:
Try it on Arduino IDE version 2

closed 2024-01-06 by thelastoutpostworkshop