thelastoutpostworkshop/gpio_viewer #24: Works but not with my code

AlxBilt, 2023-12-31:
This gpio_viewer is awesome!! but when I add my code it will not load the web-viewer. Response is refused to connect. 

#include <gpio_viewer.h> // Must me the first include in your project for ESP32 GPIO Monitoring 
GPIOViewer gpio_viewer;

// --- shield pins ---------------------

#define SHIELD_LED_PIN           26
#define SHIELD_CAN_RX            GPIO_NUM_5
#define SHIELD_CAN_TX            GPIO_NUM_4
#define SHIELD_VOLTAGE_DIVIDER   32  /* the jumper must be soldered on the v1.1 shield. Voltage divider doesn't work/exist on v1.0 */
//#define CAN_FRAME CAN.write 

// --- can ---------------------------

#include <esp32_can.h>      /* https://github.com/collin80/esp32_can */

void printFrame(CAN_FRAME *message)
{
  Serial.print(message->id, HEX);
  if (message->extended) Serial.print(" X ");
  else Serial.print(" S ");   
  Serial.print(message->length, DEC);
  for (int i = 0; i < message->length; i++) {
    Serial.print(message->data.byte[i], HEX);
    Serial.print(" ");
  }
  Serial.println();
}

void gotHundred(CAN_FRAME *frame)
{
  Serial.print("Got special frame!  ");
  printFrame(frame);
}

void setup() {
  // put your setup code here, to run once:
     Serial.begin(115200);
     
    // --- ESP32 GPIO VIEWER ---------------------------
  gpio_viewer.connectToWifi("address removed", "password removed");   // Comment the next line, If your code already include connection to Wifi
  gpio_viewer.setPort(8080);   // You can set the http port, if not set default port is 8080
   // --- ESP32 GPIO VIEWER ---------------------------
   
       Serial.println("Initializing ...");
       
  Serial.println("------------------------");
  Serial.println("    MrDIY SENDER CAN SHIELD");
  Serial.println("------------------------");
  

  CAN0.setCANPins(GPIO_NUM_5, GPIO_NUM_4);

 if(CAN0.begin(500000))
 {
  Serial.println(" CAN...............INIT");
  } else {
    Serial.println("Init Failed ...");
   } 
         Serial.println("Ready ...!");
    CAN_FRAME txFrame;
  txFrame.rtr = 0;
  txFrame.id = 0x215;
  txFrame.extended = false;
  txFrame.length = 4;
  txFrame.data.uint8[0] = 0x10;
  txFrame.data.uint8[1] = 0x1A;
  txFrame.data.uint8[2] = 0xFF;
  txFrame.data.uint8[3] = 0x5D;
  CAN0.sendFrame(txFrame);
  
  pinMode(SHIELD_LED_PIN, OUTPUT);
  digitalWrite(SHIELD_LED_PIN, LOW);
 CAN0.watchFor(); //then let everything else through anyway

 // --- ESP32 GPIO VIEWER ---------------------------
 // Must be at the end of your setup
  // gpio_viewer.setSamplingInterval(25); // You can set the sampling interval in ms, if not set default is 100ms
  gpio_viewer.begin();
}

void loop() {

byte i=0;

// CAN message frame
CAN_FRAME message;
  
  // put your main code here, to run repeatedly:
 Serial.print("Sending packet ... ");

  
  //or, just plain send traffic periodically
  
    delayMicroseconds(200);
    message.id++;
    message.length = 8;
    for(i=0;i<message.length;i++) {
     message.data.uint8[i]++;
    }
    CAN0.sendFrame(message);
  
  
  Serial.println("done");
  
   

}

thelastoutpostworkshop, 2023-12-31:
What is the IDE you are using ?
Do you get the URL to connect to the web application on the serial Monitor ?

AlxBilt, 2023-12-31:
Thank you for the response.
I am using the Arduino IDE.

Your code works fine but when I add my code to your sketch.
The both sketch code loads.
The console shows connected to the wifi address but when I try to view on
Firefox or Microsoft Edge, or Brave.
[image: image.png]

Of course with your bare code it shows.

thelastoutpostworkshop, 2023-12-31:
I will give a try to the code you provided above to see what's going on

AlxBilt, 2024-01-01:
Thank you.FYI.  When I load my code along the the GPIO viewer code, my ESP32’s blue light is awaken ( blinking) even though there is no viewer loaded in the browsers. So it must be working except for the viewer. Sent from mobile

AlxBilt, 2024-01-01:
Just a thought.My code could also be wrong. I am attempting to understand CAN communication. Sent from mobile

thelastoutpostworkshop, 2024-01-03:
Did you made any progress ?

AlxBilt, 2024-01-03:
Afternoon,

I have not been able to work on my CAN communication project.
I have been busy doing other things. I will get back to it at a later date.
Your GPIO viewer would be very helpful for me to figure out [if] the
ESP32(s)  are broadcasting and receiving data or [why] they are not.
Currently the Arduino console shows [sending packet] from Sender ESP32 but
nothing is being received on my other port for the Receiver ESP32.

I will try again later this week if I have the time.

Sent from mobile

closed 2024-01-06 by thelastoutpostworkshop