thelastoutpostworkshop/gpio_viewer #33: analogGetChannel not defined in scope

DakotaWolfi, 2024-01-02:
i tried to compile the example. but i got an error

In file included from C:\Users\Dakota\Documents\Arduino\libraries\GPIOViewer\examples\gpioviewer\gpioviewer.ino:8:0:
C:\Users\Dakota\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h: In member function 'int GPIOViewer::readGPIO(int, uint32_t*, pinTypes*)':
C:\Users\Dakota\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:293:57: error: 'analogGetChannel' was not declared in this scope
         uint8_t analogChannel = analogGetChannel(gpioNum);

i tryd to compile it using the Arduino IDE 1.8.13

DakotaWolfi, 2024-01-03:
after updating the IDE to 2.2.1
reinstalling the ESP Bords (2.0.13)
reinstaling GPIO Viewer (1.0.3)
    including dependencies(AsyncTCP1.1.4))
reinstalling AsyncEspFsWebserver (1.0.2)

when i  try to compile the provides example, with nothing else i get the following error 

In file included from C:\Users\Dakota\AppData\Local\Temp\.arduinoIDE-unsaved202403-29368-1ks9y8o.1n2m\gpioviewer\gpioviewer.ino:8:
C:\Users\Dakota\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h: In member function 'void GPIOViewer::resetStatePins()':
C:\Users\Dakota\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:206:23: error: format '%u' expects argument of type 'unsigned int', but argument 3 has type 'long unsigned int' [-Werror=format=]
         Serial.printf("GPIO View Connected, sampling interval is %ums\n", samplingInterval);
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~~~~~~~~~~
cc1plus.exe: some warnings being treated as errors

i think it has something to do with %u

thelastoutpostworkshop, 2024-01-03:
so it is a warning and your code compile ?
Some people had to revert to boards version 2.0.11 to have pins activated in the web application

DakotaWolfi, 2024-01-03:
no. it does not compile

thelastoutpostworkshop, 2024-01-03:
I saw in your code above, I think you have the wrong library :  AsyncEspFsWebserver (1.0.2) it should be : https://github.com/me-no-dev/ESPAsyncWebServer

DakotaWolfi, 2024-01-03:
i instalt that libary, but it stil gives me the same error. i dont know what im dooing wrong here

i cant paste the full error because it is... i dont know howmanny lines long, but i think this part is the interesting part

"-IC:\\Users\\Dakota\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/fb_gfx/include" "-IC:\\Users\\Dakota\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/dio_qspi/include" -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -ggdb -freorder-blocks -Wwrite-strings -fstack-protector -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu++11 -fexceptions -fno-rtti -MMD -c -Wall -Werror=all -Wextra -Os -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"doitESP32devkitV1\"" -DARDUINO_PARTITION_default -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_USB_CDC_ON_BOOT=0 "@C:\\Users\\Dakota\\AppData\\Local\\Temp\\arduino\\sketches\\9F0F6620E4B0E7534351672996E112B6/build_opt.h" "@C:\\Users\\Dakota\\AppData\\Local\\Temp\\arduino\\sketches\\9F0F6620E4B0E7534351672996E112B6/file_opts" "-IC:\\Users\\Dakota\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11\\cores\\esp32" "-IC:\\Users\\Dakota\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11\\variants\\doitESP32devkitV1" "-IC:\\Users\\Dakota\\Documents\\Arduino\\libraries\\GPIOViewer\\src" "-IC:\\Users\\Dakota\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11\\libraries\\WebServer\\src" "-IC:\\Users\\Dakota\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11\\libraries\\WiFi\\src" "-IC:\\Users\\Dakota\\Documents\\Arduino\\libraries\\ESP_Async_WebServer\\src" "-IC:\\Users\\Dakota\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11\\libraries\\FS\\src" "-IC:\\Users\\Dakota\\Documents\\Arduino\\libraries\\AsyncTCP\\src" "C:\\Users\\Dakota\\AppData\\Local\\Temp\\arduino\\sketches\\9F0F6620E4B0E7534351672996E112B6\\sketch\\gpioviewer.ino.cpp" -o "C:\\Users\\Dakota\\AppData\\Local\\Temp\\arduino\\sketches\\9F0F6620E4B0E7534351672996E112B6\\sketch\\gpioviewer.ino.cpp.o"
In file included from C:\Users\Dakota\AppData\Local\Temp\.arduinoIDE-unsaved202403-29368-1ks9y8o.1n2m\gpioviewer\gpioviewer.ino:8:
C:\Users\Dakota\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h: In member function 'void GPIOViewer::resetStatePins()':
C:\Users\Dakota\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:206:23: error: format '%u' expects argument of type 'unsigned int', but argument 3 has type 'long unsigned int' [-Werror=format=]
         Serial.printf("GPIO View Connected, sampling interval is %ums\n", samplingInterval);
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~~~~~~~~~~
cc1plus.exe: some warnings being treated as errors

Using library GPIOViewer at version 1.0.3 in folder: C:\Users\Dakota\Documents\Arduino\libraries\GPIOViewer 
Using library WebServer at version 2.0.0 in folder: C:\Users\Dakota\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WebServer 
Using library WiFi at version 2.0.0 in folder: C:\Users\Dakota\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFi 
Using library ESP Async WebServer at version 1.2.3 in folder: C:\Users\Dakota\Documents\Arduino\libraries\ESP_Async_WebServer 
Using library FS at version 2.0.0 in folder: C:\Users\Dakota\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\FS 
Using library AsyncTCP at version 1.1.4 in folder: C:\Users\Dakota\Documents\Arduino\libraries\AsyncTCP 
exit status 1

Compilation error: exit status 1

thelastoutpostworkshop, 2024-01-03:
Try this, in gpio_viewer.h, function resetStatePins(void),
change :
        Serial.printf("GPIO View Connected, sampling interval is %ums\n", samplingInterval);

to this :
        Serial.printf("GPIO View Connected, sampling interval is %lums\n", samplingInterval);

DakotaWolfi, 2024-01-03:
this workkt it compiled sucsessfuley

thelastoutpostworkshop, 2024-01-03:
Glad in now works, I fixed it and will be available in the next release.

closed 2024-01-03 by thelastoutpostworkshop