thelastoutpostworkshop/gpio_viewer #34: Board ESP32-WROOM-32D

PIBSAS, 2024-01-02:
Image: https://github.com/PIBSAS/GPView/blob/main/gpio_viewer/assets/devboards_images/ESP32-WROOM-32D.png

Pinout is the same that you have like VROOM, and this is part of same pinout for various ESP32 DevKitC: https://www.espressif.com/en/products/devkits

For the microcontrollers but with different pics:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-devkitc.html
ESP32-WROOM-32E
ESP32-WROOM-32UE(This is show in the Product selector, the rest on the docs.https://www.espressif.com/sites/default/files/dev-board/ESP32-DevKitC_S_0.png)
ESP32-WROOM-DA
ESP32-WROVER-E
ESP32-WROVER-IE
More Pics newer boards:
https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/

html for ESP32-DevKitC_S_0.png:
<div id="indicators">
  <div class="indicator" style="top: 26.666%; left: 21.6%" id="gpio36"></div>
  <div class="indicator" style="top: 30.466%; left: 21.6%" id="gpio39"></div>
  <div class="indicator" style="top: 34.266%; left: 21.6%" id="gpio34"></div>
  <div class="indicator" style="top: 38.066%; left: 21.6%" id="gpio35"></div>
  <div class="indicator" style="top: 41.866%; left: 21.6%" id="gpio32"></div>
  <div class="indicator" style="top: 45.666%; left: 21.6%" id="gpio33"></div>
  <div class="indicator" style="top: 49.466%; left: 21.6%" id="gpio25"></div>
  <div class="indicator" style="top: 53.266%; left: 21.6%" id="gpio26"></div>
  <div class="indicator" style="top: 57.066%; left: 21.6%" id="gpio27"></div>
  <div class="indicator" style="top: 60.866%; left: 21.6%" id="gpio14"></div>
  <div class="indicator" style="top: 64.666%; left: 21.6%" id="gpio12"></div>
  <div class="indicator" style="top: 72.266%; left: 21.6%" id="gpio13"></div>
  <div class="indicator" style="top: 76.066%; left: 21.6%" id="gpio9"></div>
  <div class="indicator" style="top: 79.866%; left: 21.6%" id="gpio10"></div>
  <div class="indicator" style="top: 83.666%; left: 21.6%" id="gpio11"></div>

  <div class="indicator" style="top: 22.866%; left: 78.4%" id="gpio23"></div>
  <div class="indicator" style="top: 26.666%; left: 78.4%" id="gpio22"></div>
  <div class="indicator" style="top: 30.466%; left: 78.4%" id="gpio1"></div>
  <div class="indicator" style="top: 34.266%; left: 78.4%" id="gpio3"></div>
  <div class="indicator" style="top: 38.066%; left: 78.4%" id="gpio21"></div>
  <div class="indicator" style="top: 41.866%; left: 78.4%" id="gpio23"></div>
  <div class="indicator" style="top: 45.666%; left: 78.4%" id="gpio19"></div>
  <div class="indicator" style="top: 49.466%; left: 78.4%" id="gpio18"></div>
  <div class="indicator" style="top: 53.266%; left: 78.4%" id="gpio5"></div>
  <div class="indicator" style="top: 57.066%; left: 78.4%" id="gpio17"></div>
  <div class="indicator" style="top: 60.866%; left: 78.4%" id="gpio16"></div>
  <div class="indicator" style="top: 64.666%; left: 78.4%" id="gpio4"></div>
  <div class="indicator" style="top: 68.466%; left: 78.4%" id="gpio0"></div>
  <div class="indicator" style="top: 72.266%; left: 78.4%" id="gpio2"></div>
  <div class="indicator" style="top: 76.066%; left: 78.4%" id="gpio15"></div>
  <div class="indicator" style="top: 79.866%; left: 78.4%" id="gpio8"></div>
  <div class="indicator" style="top: 83.666%; left: 78.4%" id="gpio7"></div>
  <div class="indicator" style="top: 87.466%; left: 78.4%" id="gpio6"></div>

  <div id="values">
    <div class="value value_right" style="top: 25.566%; left: -5%" id="gpio36">0</div>
    <div class="value value_right" style="top: 29.366%; left: -5%" id="gpio39">0</div>
    <div class="value value_right" style="top: 33.166%; left: -5%" id="gpio34">0</div>
    <div class="value value_right" style="top: 37.066%; left: -5%" id="gpio35">0</div>
    <div class="value value_right" style="top: 40.766%; left: -5%" id="gpio32">0</div>
    <div class="value value_right" style="top: 44.566%; left: -5%" id="gpio33">0</div>
    <div class="value value_right" style="top: 48.366%; left: -5%" id="gpio25">0</div>
    <div class="value value_right" style="top: 52.166%; left: -5%" id="gpio26">0</div>
    <div class="value value_right" style="top: 56.066%; left: -5%" id="gpio27">0</div>
    <div class="value value_right" style="top: 59.766%; left: -5%" id="gpio14">0</div>
    <div class="value value_right" style="top: 63.566%; left: -5%" id="gpio12">0</div>
    <div class="value value_right" style="top: 71.166%; left: -5%" id="gpio13">0</div>

    <div class="value" style="top: 21.766%; left: 84%" id="gpio23">0</div>
    <div class="value" style="top: 25.566%; left: 84%" id="gpio22">0</div>
    <div class="value" style="top: 29.366%; left: 84%" id="gpio1">0</div>
    <div class="value" style="top: 33.166%; left: 84%" id="gpio3">0</div>
    <div class="value" style="top: 37.066%; left: 84%" id="gpio21">0</div>
    <div class="value" style="top: 44.566%; left: 84%" id="gpio19">0</div>
    <div class="value" style="top: 48.366%; left: 84%" id="gpio18">0</div>
    <div class="value" style="top: 52.166%; left: 84%" id="gpio5">0</div>
    <div class="value" style="top: 56.066%; left: 84%" id="gpio17">0</div>
    <div class="value" style="top: 59.766%; left: 84%" id="gpio16">0</div>
    <div class="value" style="top: 63.566%; left: 84%" id="gpio4">0</div>
    <div class="value" style="top: 67.366%; left: 84%" id="gpio0">0</div>
    <div class="value" style="top: 71.166%; left: 84%" id="gpio2">0</div>
    <div class="value" style="top: 75.066%; left: 84%" id="gpio15">0</div>
  </div>
  <div class="stats" style="top: 58%; left: 30%" id="freeHeap">Free Heap: 142.10 KB</div>
  <div class="stats" style="top: 64%; left: 30%" id="freeRAM">Free Sketch: 1.25 MB</div>
</div>

Thanks for library, i've use [Color Picker online](https://redketchup.io/color-picker) to get pixels on PNG then look for porcentage, maybe is not the smartest way he, but in the image of my board ESP32-WROOM-32D the dots are more oval and i dont know how to fix that. Hope soon your library be in Library Manager, nice work!

thelastoutpostworkshop, 2024-01-02:
What is the board exactly you want me to add ?

PIBSAS, 2024-01-02:
Oh sorry I add more just to share. The ESP32-WROOM-32D

thelastoutpostworkshop, 2024-01-02:
Currently the 30 pins and 38 pins version are available, yours is different ?

PIBSAS, 2024-01-02:
38 pins, the vroom bad name, is in fact the ESP32-WROOM, not the ESP32-WROOM-32D

closed 2024-01-04 by PIBSAS