thelastoutpostworkshop/gpio_viewer #36: basic functionality doesn't work for me

danny-water, 2024-01-02:
Hi, 
Looks like awsome project, tried it, I see the web page but no blinking led, all gpios are constatntly off.
this is my code:

/***
This example is intended to demonstrate the use of the GPIO Viewer Library.

Tutorial : https://youtu.be/UxkOosaNohU
Documentation : https://github.com/thelastoutpostworkshop/gpio_viewer
***/

#include <gpio_viewer.h> // Must me the first include in your project
GPIOViewer gpio_viewer;

void setup()
{
  Serial.begin(115200);

  // Comment the next line, If your code aleady include connection to Wifi
  gpio_viewer.connectToWifi("Tigo", "Tigo!1234");
  // gpio_viewer.setPort(5555);                 // You can set the http port, if not set default port is 8080

  // Your own setup code start here
  pinMode(2, OUTPUT);
  digitalWrite(2, HIGH); // turn the LED on

  // Must be at the end of your setup
  // gpio_viewer.setSamplingInterval(25); // You can set the sampling interval in ms, if not set default is 100ms
  gpio_viewer.begin();

  
}

// You don't need to change your loop function
void loop() {
  
  digitalWrite(2, HIGH); // turn the LED on
  delay(500);             // wait for 500 milliseconds
  digitalWrite(2, LOW);  // turn the LED off
  delay(500);  
}
// The rest of your code here

[image]

- BTW - I see the LED blinking on the board, anything obvious i'm missing ?

thelastoutpostworkshop, 2024-01-02:
What is the URL you get on the Serial Monitor ?

danny-water, 2024-01-02:
http://192.168.1.93:8080

thelastoutpostworkshop, 2024-01-02:
I don't have this board, but it should work.  Have you make it work with another board ?
If you use the generic view, do you see pin 2 blink on the application ?

danny-water, 2024-01-02:
not sure I have another board, will try to find.
unfortunately I don't see pin 2 blink on the generic board as well

thelastoutpostworkshop, 2024-01-02:
if you refresh your browser, you should see a message on the Serial monitor indicating that the ESP32 is connected to the web application : 
[image]

danny-water, 2024-01-02:
yeah, I see that every time I refresh.
but also I see 0 channels are used .
I'm using a different chip, same thing.

I'm using arduino IDE version 1.8.19 - maybe that the problem ?

Here is the printout from the serial monitor:
================================
Chip Model: ESP32-S3
Chip version: 0
Numer of cores: 2
Flash Chip Size: 8388608
Flash Chip Speed: 80000000
================================
0 pins are PWM
0 channels are used
GPIOView Web Application URL is: http://192.168.1.93:8080
GPIO View Connected, sampling interval is 100ms
GPIO View Connected, sampling interval is 100ms
GPIO View Connected, sampling interval is 100ms

thelastoutpostworkshop, 2024-01-02:
yes if possible try it on version Arduino 2

thelastoutpostworkshop, 2024-01-02:
other question, are you seeing Free Heap memory changing ?

danny-water, 2024-01-03:
The heap was not changing as well.
tried the arduino 2 ide and it works great !!!

[image]

Thank you so much for your great and quick responses !!

thelastoutpostworkshop, 2024-01-03:
Glad it's working!

closed 2024-01-03 by thelastoutpostworkshop