thelastoutpostworkshop/gpio_viewer #38: GPIO viewer does not work

Marcelverhoeven, 2024-01-02:
I installed the latest release of GPOI viewer (1,0,3)
I installed https://github.com/me-no-dev/ESPAsyncWebServer
I entered in the gpioviewer.ino (= example code)  my SSID and Pass
I have added code:
 In the setup: pinMode(4, OUTPUT); and digitalWrite(4, LOW);
 In the loop: digitalWrite(4, HIGH); delay(1000); digitalWrite(4, LOW);  delay(1000);

The Led connected to GPIO is flashing  as expected at the rate of 1 sec,
I also restated Arduino,

On my browser I see the Picture of the ESP DevKit v1 but ALL pins remain at "0"
What might be the issue???

thelastoutpostworkshop, 2024-01-02:
What view are you using in the web application ?

Marcelverhoeven, 2024-01-02:
[image]

I also tried several other views and port 8080 and 5555

still the same issue

thelastoutpostworkshop, 2024-01-02:
if you refresh your browser, you should see a message on the Serial monitor indicating that the ESP32 is connected to the web application :
[image]

Marcelverhoeven, 2024-01-02:
That is correct:
[image]

Marcelverhoeven, 2024-01-02:
[image]

I also tried port 
8080 and 5555

Marcelverhoeven, 2024-01-02:
Note: Free heap and Free Sketch are alive
[image]

But no GPIO info

thelastoutpostworkshop, 2024-01-02:
Try clearing your browser cache data.
Also, do you see the value of free heap changing ?

Marcelverhoeven, 2024-01-02:
Try clearing your browser cache data ==> DONE
Also, do you see the value of free heap changing ==> YES
No solution,

thelastoutpostworkshop, 2024-01-02:
that is strange, have you the latest version of Espressif boars : 
[image]

wazzaguy, 2024-01-02:
I was experiencing the same problem but got it to work.
I had to downgrade the board from version 2.0.14 to 2.0.11 and recompile.
[image]

thelastoutpostworkshop, 2024-01-02:
Thanks for you help @wazzaguy, I am going to do some test with version 2.0.14

thelastoutpostworkshop, 2024-01-02:
Just tested the library against Espressif Esp32 boars v.2.0.14 (latest version) and everything is working

wazzaguy, 2024-01-03:
I updated from 2.0.11 to 2.0.14 and its not working again.
@Marcelverhoeven does downgrading to 2.0.11 work for you?

@thelastoutpostworkshop which Board Manager URL are you using?
I'm using https://espressif.github.io/arduino-esp32/package_esp32_index.json

thelastoutpostworkshop, 2024-01-03:
@wazzaguy I have used the boards manager in the Arduino IDE 2 : 
[image]

thelastoutpostworkshop, 2024-01-03:
This is the library : https://github.com/espressif/arduino-esp32

whilexx, 2024-01-03:
This Worked for me as well. The Board version is the Problem 2.0.11 works fine. BTW, great tool. Thx.

KwangYeonCHO, 2024-01-03:
I also encountered the same problem, web server, development board images, IO not working, always in 0.  [ VSCODE PLATFORM]

labixlerOSL, 2024-01-03:
I had the same issue. I rolled back to 2.0.11 and it works.

thelastoutpostworkshop, 2024-01-03:
I am kind of thinking that the reinstallation of the ESP32 boards solve the problem, if someone could try to go back to 2.0.14 to test it again please.

thelastoutpostworkshop, 2024-01-03:
Thanks for trying it, I will have to investigate further.  As for the activities, this could be "floating pins", the library reads the actual value of the pins

Lynetk, 2024-01-03:
> I updated to 2.0.14 and it was no longer working. I have reverted to 2.0.11 and it is now showing something again. I have noticed, however, that there seems to be a lot of activity for a very basic program. All I am doing at the moment is flashing a single LED, but 14 other pins are randomly switching state.

Same "issue" for me. 2.0.14 not working with s2 mini board. Reverting to 2.0.11 gets it going, but there is a lot of activity on non-used pins and after about 20-30 seconds the board resets and it takes some time to get back up. This pattern repeats itself.

I also run a simple blink program and use the LOLIN S2 Mini

```
GPIOViewer gpio_viewer;
#define LED_GPIO 1

void setup()
{
  Serial.begin(115200);
  gpio_viewer.connectToWifi("removed", "removed");
  pinMode(LED_GPIO, OUTPUT);
  gpio_viewer.begin();
}

void loop() {
  digitalWrite(LED_GPIO, HIGH);
  delay(1000);
  digitalWrite(LED_GPIO, LOW);
  delay(1000);  
}
```

Marcelverhoeven, 2024-01-03:
I installed 2,0,11 and it works!
This is really a great library!!!
I am not strong enough to find out why 2,0,13 does not work ==> It would be nice if someone could put his teeth in it.
I will recommend it to my friends.
[image]

thelastoutpostworkshop, 2024-01-03:
Glad it's working now

closed 2024-01-03 by thelastoutpostworkshop