thelastoutpostworkshop/gpio_viewer #42: Code works not with WIFI AP / is it Possible ?

HolzCoder, 2024-01-03:
Hello together, 

I implementet the viewer in my Code and it works. 
But my Projekt has an Wifi Access Point. So i Don#t connect the an WIFI i make my own. 
I can reach then the Viewer but it dont show me the states of the Pins.
I think the Problem is that i have an AP. 

Can someone help me ?

Console Log: 

18:38:04.734 -> Setting AP (Access Point)…AP IP address: 192.168.4.1
18:38:04.843 -> ESP1 added
18:38:04.843 -> ESP2 added
18:38:04.843 -> ESP3 added
18:38:04.843 -> ESP4 added
18:38:04.843 -> HTTP server started
18:38:04.843 -> 0 pins are PWM
18:38:04.843 -> 0 channels are used
18:38:04.843 -> ESP32 is not connected to WiFi.
18:38:04.843 -> Aktive Esps: 4

Sorry for my bad English.

thelastoutpostworkshop, 2024-01-03:
I will have to modify the library to support this, can you give me an example on how you setup an access point, so I can test it on my side

HolzCoder, 2024-01-03:
Yes, I have created a programme here with sections from my program which is roughly the same as mine. 
I have attached the programme.

[GPIOViewerWithAP.zip](https://github.com/thelastoutpostworkshop/gpio_viewer/files/13823376/GPIOViewerWithAP.zip)

thelastoutpostworkshop, 2024-01-03:
I am going to give it a try to see how to make the library compatible

thelastoutpostworkshop, 2024-01-04:
when I try your code without GPIOViewer library, I am unable to connect to the server using the softAPIP address printed, which is 192.168.4.1 on my network.  Normally my esp32 connect to 192.168.1.xxx when using WIFI_STA

I don't know very well the WIFI_AP_STA mode, is there anything I am missing ?

thelastoutpostworkshop, 2024-01-04:
I just read that 192.168.4.1 is the default, I can change it with  IPAddress local_IP(192, 168, 1, 1);
  IPAddress gateway(192, 168, 1, 1);
  IPAddress subnet(255, 255, 255, 0);

thelastoutpostworkshop, 2024-01-04:
ok now I understand how it works, I was able to connect to the Wifi of the ESP32.
But it will not work with the GPIOViewer be cause it requires internet access, the web application is hosted on Github pages.

thelastoutpostworkshop, 2024-01-05:
Internet Sharing (Not Straightforward): For devices connected to the ESP32's AP to access the internet, the ESP32 must be configured to route traffic between its Station interface and AP interface. This requires setting up NAT or similar routing configurations, which is quite complex and not directly supported by standard Arduino libraries.

Do you  have internet access with your ESP32 setup as an access point ?

HolzCoder, 2024-01-05:
Hey, thanks for the answer. 

ok i didn't realise that the ESP needed internet access. my system is independent and self-contained without internet access. The system should work stand alone. Then it probably doesn't work with the GPIO Viewer. but thanks for your efforts anyway.

closed 2024-01-05 by thelastoutpostworkshop