thelastoutpostworkshop/gpio_viewer #48: GPIOViewer v1.0.5

Marcelverhoeven, 2024-01-05:
I am sorry to say: earlier this week it worked perfectly (using ESP Board 2,0,11.
As a test I left the power on for 24hours.

Now i get this:
[image]

I have updated the library to 1,0,5: but the same result.

My test code is:
const int ledPin = 2; 
#include <gpio_viewer.h> // Must me the first include in your project
GPIOViewer gpio_viewer;

void setup() {
  Serial.begin(115200);
  // Comment the next line, If your code already include connection to Wifi
  gpio_viewer.connectToWifi("************************", "*******************************");
  gpio_viewer.setPort(5555);  
  pinMode(ledPin, OUTPUT); // Initialize GPIO 2 as an output
  gpio_viewer.setSamplingInterval(250);
  gpio_viewer.begin();
}

void loop() {
  digitalWrite(ledPin, HIGH); // Turn on the LED
  delay(1000); // Wait for 1 second

  digitalWrite(ledPin, LOW); // Turn off the LED
  delay(1000); // Wait for 1 second
}
I am very fond to integrate this GPIOVIEWER in many of my projects but I must be sure that it is stable,

Marcelverhoeven, 2024-01-05:
I am also confused: On the Github it is said "Latest Version" = 1.0.5 but when I install that it seems to be 1.0.4!!!

KwangYeonCHO, 2024-01-05:
[image]

1.0.5 I also encountered the same problem, the webpage cannot be displayed and remains in an incomplete open state

thelastoutpostworkshop, 2024-01-05:
Is there a message on the Serial Monitor like "GPIOViewer >> Connected, sampling interval is" that appears when you refresh your browser window ?

closed 2024-01-06 by thelastoutpostworkshop