thelastoutpostworkshop/gpio_viewer #50: ESPHome functionality

JoeAverge9, 2024-01-05:
Would it be possible to enable this within the ESPHome project?
https://www.esphome.io/

JoeAverge9, 2024-01-05:
I also found a similar request going the same way. Not sure where the best place for this request should be.
https://github.com/esphome/feature-requests/issues/2546

thelastoutpostworkshop, 2024-01-05:
I don't know ESPHome.  Are you creating a sketch when working with ESPHome ?

JoeAverge9, 2024-01-05:
> I don't know ESPHome. Are you creating a sketch when working with ESPHome ?

ESPHome is a declarative framework for programing ESP32 and ESP8266 dev boards

thelastoutpostworkshop, 2024-01-05:
Can you import libraries ?

JoeAverge9, 2024-01-05:
> Can you import libraries ?

I do not know for sure but depending on what you declare in the YAML config I believe this happens during the compile process before upload.

thelastoutpostworkshop, 2024-01-05:
I will have a look at their documentation, if they don't allow third party libraries, there is not much I will able to do

trantoriana, 2024-01-07:
ESPHome does  allow importing external components using the 'external' component:

* https://esphome.io/components/external_components.html

@thelastoutpostworkshop  : Would be great if I knew what to import.

I am willing to test this and contribute to documenting.

T.

thelastoutpostworkshop, 2024-01-07:
Thanks a lot for the info and sure I will need help for testing.

trantoriana, 2024-01-07:
It looks like the source needs to be rearrange in a specific format to allow ESPHome to import such a 'component', as per:

* https://esphome.io/guides/contributing#contributing-to-esphome

Perhaps core developers of ESPHome are willing to contribute turning GPIOViewer into such a component.

Fingers crossed

thelastoutpostworkshop, 2024-01-07:
Yes and some people open issues on it, on the ESPHome repository

thelastoutpostworkshop, 2024-01-07:
It appears, that IDF is needed for ESPHome, currently GPIOViewer library is based on the Arduino Framework.  So a port to IDF is needed, I do not know at this point the amount of work this port will required.

trantoriana, 2024-01-07:
If i'd only knew what this means, but it seems possible to create a arduino based library as component in IDF.

At least, that is what it looks like they done here:

* https://github.com/sukesh-ak/Arduino-Library-with-IDF

Perhaps a similar method can be used here ??

thelastoutpostworkshop, 2024-01-07:
Thanks I will look into it

oka4henry, 2024-01-16:
> Can you import libraries ?

Yes

oka4henry, 2024-01-16:
Yes, you can.

platte-73, 2024-01-17:
So, is it possible to use GPIO Viewer with ESPHome?

thelastoutpostworkshop, 2024-01-17:
I think so, if someone wants to submit PRs for it, you are welcomed.

tache, 2024-02-06:
First, this is an awesome capability.
Second, I am +1 for the functionality being on ESPHome. Has anyone started the effort yet? I am on board to help.

thelastoutpostworkshop, 2024-02-06:
Nobody submitted PRs for this yet, your are welcome to contribute.

tache, 2024-02-07:
Going to give it a go, but got to get up to speed on ESPHome component development and then off to the races.

thelastoutpostworkshop, 2024-02-07:
Thanks, I do not know how much of a learning curve it is.

erazor666, 2024-02-11:
I really like this functionality and I really hope someone makes it come to pass. Not much good on coding mysel sadly 🤔

Joo01, 2024-03-06:
Yes, GPIOViewer in ESPHome would be a perfect addition!
Hope someone gets it working :-)

Jaharmi, 2024-03-10:
This looks great, just found it and am also interested in the possibility with ESPHome. In the meantime, for a particular problem I’m having, I was thinking I’d temporarily re-flash the affected device. It’s not like the config for that device will go away in ESPHome.

Jaharmi, 2024-03-11:
I was able to add this to ESPHome.

```
esphome:
  libraries:
    # A library from PlatformIO
    - thelastoutpostworkshop/GPIOViewer
```

However, I don’t think this results in the “#include” being set and the Web server running on port 8080. I may be wrong, as the ESPHome device I was testing already had a Web server on port 80. (This also meant it had the other required libraries installed.)

MrC0nst, 2024-04-22:
> I was able to add this to ESPHome.
> 
> ```
> esphome:
>   libraries:
>     # A library from PlatformIO
>     - thelastoutpostworkshop/GPIOViewer
> ```

So you have this addon working in esphome just with 4 lines of code only?
I cant reach 80 or 8080 port of my esp to view gpio states

MrC0nst, 2024-04-22:
This example works only with inpit pins https://github.com/dvb6666/esphome-components/blob/main/examples/test-gpio-viewer.yaml

wacsy, 2025-04-29:
The example from dvb6666 doesn't work for me. It failed the compilation. but the following is good for compilation and OTA:

```
esphome:
  name: e32s2
  friendly_name: gproviewer
  libraries:
    - thelastoutpostworkshop/GPIOViewer
 external_components:
  - source: github://dvb6666/esphome-components
    refresh: 1sec
gpio_viewer:
  port: 82 # default 8080
  sampling_interval: 200 # default 100
web_server:
  port: 80
  version: 3
  local: true
```
Then I can visit it at port 82, but it doesn't give any meaningful information so far.

GPIOViewer is working great with Tasmota on the other hand

chrisjanbothma79, 2025-10-26:
INFO ESPHome 2025.10.3
INFO Reading configuration g:\esphome\new1.yaml...
INFO Updating https://github.com/dvb6666/esphome-components.git@None
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Joamon\AppData\Local\Programs\Python\Python311\Scripts\esphome.exe\__main__.py", line 6, in <module>
  File "C:\Users\Joamon\AppData\Local\Programs\Python\Python311\Lib\site-packages\esphome\__main__.py", line 1387, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Joamon\AppData\Local\Programs\Python\Python311\Lib\site-packages\esphome\__main__.py", line 1374, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Joamon\AppData\Local\Programs\Python\Python311\Lib\site-packages\esphome\__main__.py", line 698, in command_compile
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "C:\Users\Joamon\AppData\Local\Programs\Python\Python311\Lib\site-packages\esphome\__main__.py", line 442, in write_cpp
    generate_cpp_contents(config)
  File "C:\Users\Joamon\AppData\Local\Programs\Python\Python311\Lib\site-packages\esphome\__main__.py", line 454, in generate_cpp_contents
    CORE.flush_tasks()
  File "C:\Users\Joamon\AppData\Local\Programs\Python\Python311\Lib\site-packages\esphome\core\__init__.py", line 779, in flush_tasks
    self.event_loop.flush_tasks()
  File "C:\Users\Joamon\AppData\Local\Programs\Python\Python311\Lib\site-packages\esphome\coroutine.py", line 341, in flush_tasks
    next(task.iterator)
  File "C:\Users\Joamon\AppData\Local\Programs\Python\Python311\Lib\site-packages\esphome\__main__.py", line 431, in wrapped
    await coro(conf)
  File "C:\Users\Joamon\AppData\Local\Programs\Python\Python311\Lib\site-packages\esphome\core\config.py", line 482, in to_code
    cg.add_library(None, None, lib)
  File "C:\Users\Joamon\AppData\Local\Programs\Python\Python311\Lib\site-packages\esphome\cpp_generator.py", line 591, in add_library
    CORE.add_library(Library(name, version, repository))
  File "C:\Users\Joamon\AppData\Local\Programs\Python\Python311\Lib\site-packages\esphome\core\__init__.py", line 824, in add_library
    raise ValueError(f"The library for {library.repository} must have a name")
ValueError: The library for https://github.com/dvarrel/AsyncTCP.git must have a name
sys:1: RuntimeWarning: coroutine 'to_code' was never awaited
sys:1: RuntimeWarning: coroutine '_add_platform_defines' was never awaited
sys:1: RuntimeWarning: coroutine '_add_automations' was never awaited

closed 2025-11-29 by thelastoutpostworkshop