thelastoutpostworkshop/gpio_viewer #53: Question: Will you be allowing GitHub Pull Requests

toetoast, 2024-01-05:
Do you have any plans to allow external developers the ability to contribute to this Repo by way of PRs?

Thanks
Tony

thelastoutpostworkshop, 2024-01-05:
No plans, but you are welcomed to submit PRs

toetoast, 2024-01-05:
Ok cool. I did try earlier but got this message:

PS R:\> git push --set-upstream origin "Keywords_Update_Methods"
info: please complete authentication in your browser...
remote: Permission to thelastoutpostworkshop/gpio_viewer.git denied to toetoast.
fatal: unable to access 'https://github.com/thelastoutpostworkshop/gpio_viewer.git/': The requested URL returned error: 403

thelastoutpostworkshop, 2024-01-05:
You have to fork the repo

thelastoutpostworkshop, 2024-01-05:
for now I prefer this way to get PRs : https://docs.github.com/en/get-started/exploring-projects-on-github/contributing-to-a-project

closed 2024-01-05 by thelastoutpostworkshop