thelastoutpostworkshop/gpio_viewer #54: Noob error on example startup/verify?

argrithmag, 2024-01-05:
I'm following your tutorial, and I have an Arduino uno.  When I try to just verify the sample code, I get the following:
I installed the 1.06, ESP Webserver, and the AsyncTCP.
```
In file included from C:\Users\XXXXXX\OneDrive\Documents\Arduino\gpioviewer\gpioviewer.ino:8:0:
C:\Users\XXXXXX\OneDrive\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:5:10: fatal error: WebServer.h: No such file or directory
 #include <WebServer.h>
          ^~~~~~~~~~~~~
compilation terminated.

exit status 1
```

argrithmag, 2024-01-05:
I realized that nano is not an esp32....  like I said, noob to all this.

closed 2024-01-05 by argrithmag

thelastoutpostworkshop, 2024-01-05:
Only ESP32 boards are supported, the Nano is not, but there is a Nano ESP32 if you want to try ESP32,

argrithmag, 2024-01-05:
Thanks!  I wasn't expecting a reply.  I'll look for one of those.