thelastoutpostworkshop/gpio_viewer #55: Is it possible to host the code locally?

happytm, 2024-01-06:


thelastoutpostworkshop, 2024-01-06:
Yes it could be possible, you need a web server setup locally on the same network as the ESP32 to serve the assets of the web application.  However, currently the URL to get the assets are hardcoded in the library.  I will consider it for a future release as well as instructions on how to do it.  I need to do some testing to see how it could work out.

thelastoutpostworkshop, 2024-01-06:
However, serving the assets from the ESP32 itself is not possible, the assets are too large, right now they are at 26MB.

happytm, 2024-01-06:
OK. understood. Thanks.

wulph111, 2024-05-07:
I would like to see a compile-time option to select the board resources and host the (lower-resolution?) resources for that particular device inside the ESP32 filesystem.

B4ckZ, 2025-05-07:
> However, serving the assets from the ESP32 itself is not possible, the assets are too large, right now they are at 26MB.

Hi ! I'm completely new to this, but when you talk about assets, are you referring to the photos on the boards? If so, why not convert the board design to vector or SVG format to reduce weight? Or perhaps I didn't quite understand what you meant?

P.-S., I've just come across your project on YouTube and it's going to be a great help in diagnosing the next bugs I encounter.

Thank you so much for all your hard work and dedication. It's much appreciated.

thelastoutpostworkshop, 2025-05-07:
Assets are not only board images, but the web application itself.  The web application will never fit on an ESP32 and cannot run, it's using javascript.

B4ckZ, 2025-05-07:
> Assets are not only board images, but the web application itself. The web application will never fit on an ESP32 and cannot run, it's using javascript.

Oh right, I knew it couldn't be that simple ^^ thanks for the additional information.

saharesp, 2025-12-11:
Здравствуйте! Если это еще актуально: у меня веб-приложение расположено на ESP-32  и используется JavaScript. Готов объяснить как решить проблему.