thelastoutpostworkshop/gpio_viewer #56: Support for EPS32S2-Wrover missing

K-Lab-KK, 2024-01-06:
Hi, 
unfortunately the board ESP32S2-Wrover is missing.

It will be very helpful,if this will be available ..

Thanks 
Konny

thelastoutpostworkshop, 2024-01-06:
Do you have a specific board model you would like ?

CharlesGodwin, 2024-01-06:
I just hit the same problem using VS Code and PlatformIO. I'm using[ this board](https://thingpulse.com/product/epulse-low-power-esp32-development-board/) and this is is my platformio.ini
<pre>
[platformio]
default_envs = esp-wrover-kit

[env:esp-wrover-kit]
; https://thingpulse.com/product/epulse-low-power-esp32-development-board/
platform = espressif32
board = esp-wrover-kit
framework = arduino
lib_deps =
    https://github.com/dvarrel/AsyncTCP.git
    https://github.com/me-no-dev/ESPAsyncWebServer.git
    https://github.com/thelastoutpostworkshop/gpio_viewer.git</pre>

I get these compile errors
<pre>
.pio\libdeps\esp-wrover-kit\GPIOViewer\src/gpio_viewer.h: In member function 'void GPIOViewer::begin()':
.pio\libdeps\esp-wrover-kit\GPIOViewer\src/gpio_viewer.h:94:70: error: 'ESP_ARDUINO_VERSION_MAJOR' was not declared in this scope
         Serial.printf("GPIOViewer >> ESP32 Core Version %d.%d.%d\n", ESP_ARDUINO_VERSION_MAJOR, ESP_ARDUINO_VERSION_MINOR, ESP_ARDUINO_VERSION_PATCH);   
                                                                      ^
.pio\libdeps\esp-wrover-kit\GPIOViewer\src/gpio_viewer.h:94:97: error: 'ESP_ARDUINO_VERSION_MINOR' was not declared in this scope
         Serial.printf("GPIOViewer >> ESP32 Core Version %d.%d.%d\n", ESP_ARDUINO_VERSION_MAJOR, ESP_ARDUINO_VERSION_MINOR, ESP_ARDUINO_VERSION_PATCH);   
                                                                                                 ^
.pio\libdeps\esp-wrover-kit\GPIOViewer\src/gpio_viewer.h:94:124: error: 'ESP_ARDUINO_VERSION_PATCH' was not declared in this scope
         Serial.printf("GPIOViewer >> ESP32 Core Version %d.%d.%d\n", ESP_ARDUINO_VERSION_MAJOR, ESP_ARDUINO_VERSION_MINOR, ESP_ARDUINO_VERSION_PATCH);   
                                                                                                                            ^
.pio\libdeps\esp-wrover-kit\GPIOViewer\src/gpio_viewer.h: In member function 'int GPIOViewer::readGPIO(int, uint32_t*, pinTypes*)':
.pio\libdeps\esp-wrover-kit\GPIOViewer\src/gpio_viewer.h:328:57: error: 'analogGetChannel' was not declared in this scope
         uint8_t analogChannel = analogGetChannel(gpioNum);
                                                         ^
</pre>

thelastoutpostworkshop, 2024-01-06:
it's related to plateformIO, please open a new issue for this one

CharlesGodwin, 2024-01-06:
> it's related to plateformIO, please open a new issue for this one

I deleted my comment. You may want to delete this one.

closed 2024-01-08 by thelastoutpostworkshop