thelastoutpostworkshop/gpio_viewer #57: Feature Request: Adding a Tick Box called "Show Legend" to the webpage

toetoast, 2024-01-06:
What do you think about adding a new feature to the webpage that is enabled via a tick box called "Show Legend".
Once ticked, this would make the selected ESP32 "Board Legend" visible on the page.

The advantage with this feature is that you can see more details about the function of each GPIO alongside the current status.
I've made a mock-up screen shown here to simulate what this could look like.

For most of the supported boards, the legend is in the Repo anyway, so I thought this would make sense to utilise the asset.

Cheers
Tony

[image]

thelastoutpostworkshop, 2024-01-06:
Greate Idea!

toetoast, 2024-01-06:
No worries. Happy to help where I can.

thelastoutpostworkshop, 2024-08-26:
I can detect touch and ADC pins, but no other pin functions, there is no API in the ESP32 core to get that information.
I have submitted it to Espressif.
As of now you can see in the web application pins that have touch functions or ADC functions.
To see which one, you have to click on a pin for now.

closed 2024-08-26 by thelastoutpostworkshop