thelastoutpostworkshop/gpio_viewer #58: Problem building with PlatformIO and ESP32

CharlesGodwin, 2024-01-06:
I just hit the same problem using VS Code and PlatformIO. I'm using[ this board](https://thingpulse.com/product/epulse-low-power-esp32-development-board/) and this is is my platformio.ini
<pre>
[platformio]
default_envs = esp-wrover-kit

[env:esp-wrover-kit]
; https://thingpulse.com/product/epulse-low-power-esp32-development-board/
platform = espressif32
board = esp-wrover-kit
framework = arduino
lib_deps =
    https://github.com/dvarrel/AsyncTCP.git
    https://github.com/me-no-dev/ESPAsyncWebServer.git
    https://github.com/thelastoutpostworkshop/gpio_viewer.git</pre>

I get these compile errors
<pre>
.pio\libdeps\esp-wrover-kit\GPIOViewer\src/gpio_viewer.h: In member function 'void GPIOViewer::begin()':
.pio\libdeps\esp-wrover-kit\GPIOViewer\src/gpio_viewer.h:94:70: error: 'ESP_ARDUINO_VERSION_MAJOR' was not declared in this scope
         Serial.printf("GPIOViewer >> ESP32 Core Version %d.%d.%d\n", ESP_ARDUINO_VERSION_MAJOR, ESP_ARDUINO_VERSION_MINOR, ESP_ARDUINO_VERSION_PATCH);   
                                                                      ^
.pio\libdeps\esp-wrover-kit\GPIOViewer\src/gpio_viewer.h:94:97: error: 'ESP_ARDUINO_VERSION_MINOR' was not declared in this scope
         Serial.printf("GPIOViewer >> ESP32 Core Version %d.%d.%d\n", ESP_ARDUINO_VERSION_MAJOR, ESP_ARDUINO_VERSION_MINOR, ESP_ARDUINO_VERSION_PATCH);   
                                                                                                 ^
.pio\libdeps\esp-wrover-kit\GPIOViewer\src/gpio_viewer.h:94:124: error: 'ESP_ARDUINO_VERSION_PATCH' was not declared in this scope
         Serial.printf("GPIOViewer >> ESP32 Core Version %d.%d.%d\n", ESP_ARDUINO_VERSION_MAJOR, ESP_ARDUINO_VERSION_MINOR, ESP_ARDUINO_VERSION_PATCH);   
                                                                                                                            ^
.pio\libdeps\esp-wrover-kit\GPIOViewer\src/gpio_viewer.h: In member function 'int GPIOViewer::readGPIO(int, uint32_t*, pinTypes*)':
.pio\libdeps\esp-wrover-kit\GPIOViewer\src/gpio_viewer.h:328:57: error: 'analogGetChannel' was not declared in this scope
         uint8_t analogChannel = analogGetChannel(gpioNum);
                                                         ^
</pre>

rescalle83, 2024-01-06:
Newbie here and getting the following errors when trying to compile
In file included from C:\Users\rober\AppData\Local\Temp\.arduinoIDE-unsaved202406-12528-1d9cd3.mz2qk\gpioviewer\gpioviewer.ino:34:
C:\Users\rober\OneDrive\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:11:10: fatal error: ESPAsyncWebServer.h: No such file or directory
   11 | #include <ESPAsyncWebServer.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1

thelastoutpostworkshop, 2024-01-06:
I need to see your platform.ini content

rescalle83, 2024-01-06:
I am using Arduino

rescalle83, 2024-01-06:
I have now tried everything I can think of. I am using Arduino latest build and can compile Blynk apps and other apps with no issues. When trying to compile the GPIOViewer example or simply add the lines manually, I get the same error every time.

 

In file included from C:\Users\rober\OneDrive\Documents\Arduino\libraries\GPIOViewer\gpioviewer\gpioviewer.ino:8:

c:\Users\rober\OneDrive\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:11:10: fatal error: ESPAsyncWebServer.h: No such file or directory

   11 | #include <ESPAsyncWebServer.h>

      |          ^~~~~~~~~~~~~~~~~~~~~

compilation terminated.

 

exit status 1

 

Compilation error: exit status 1

 

Below is my .ini file

 

;PlatformIO Project Configuration File

;

;   Build options: build flags, source filter

;   Upload options: custom upload port, speed and extra flags

;   Library options: dependencies, extra library storages

;   Advanced options: extra scripting

;

; Please visit documentation for the other options and examples

; https://docs.platformio.org/page/projectconf.html

 

[platformio]

; ============================================================

; chose environment:

; ESP8266

; ESP32

; ============================================================

default_envs = ESP8266

;default_envs = ESP32

 

[env]

; ============================================================

; Serial configuration

; choose upload speed, serial-monitor speed

; ============================================================

upload_speed = 921600

;upload_port = COM11

;monitor_speed = 9600

;monitor_port = COM11

 

; Checks for the compatibility with frameworks and dev/platforms

lib_compat_mode = strict

lib_ldf_mode = chain+

;lib_ldf_mode = deep+

 

lib_deps =

; PlatformIO 4.x

;  LittleFS_esp32@>=1.0.6

; PlatformIO 5.x 

; lorol/LittleFS_esp32@>=1.0.6

 

 

build_flags =

; set your debug output (default=Serial)

-D DEBUG_ESP_PORT=Serial

; comment the folowing line to enable WiFi debugging

-D NDEBUG

 

[env:ESP8266]

platform = espressif8266

framework = arduino

; ============================================================

; Board configuration

; choose your board by uncommenting one of the following lines

; ============================================================

;board = gen4iod

;board = huzzah

;board = oak

;board = esp_wroom_02

;board = espduino

;board = espectro

;board = espino

;board = espresso_lite_v1

;board = espresso_lite_v2

;board = esp12e

;board = esp01_1m

;board = esp01

;board = esp07

;board = esp8285

;board = heltec_wifi_kit_8

;board = inventone

;board = nodemcu

board = nodemcuv2

;board = modwifi

;board = phoenix_v1

;board = phoenix_v2

;board = sparkfunBlynk

;board = thing

;board = thingdev

;board = esp210

;board = espinotee

;board = d1

;board = d1_mini

;board = d1_mini_lite

;board = d1_mini_pro

;board = wifi_slot

;board = wifiduino

;board = wifinfo

;board = wio_link

;board = wio_node

;board = xinabox_cw01

;board = esp32doit-devkit-v1

 

[env:ESP32]

platform = espressif32

framework = arduino

; ============================================================

; Board configuration

; choose your board by uncommenting one of the following lines

; ============================================================

;board = esp32cam

;board = alksesp32

;board = featheresp32

;board = espea32

;board = bpi-bit

;board = d-duino-32

board = esp32doit-devkit-v1

;board = pocket_32

;board = fm-devkit

;board = pico32

;board = esp32-evb

;board = esp32-gateway

;board = esp32-pro

;board = esp32-poe

;board = oroca_edubot

;board = onehorse32dev

;board = lopy

;board = lopy4

;board = wesp32

;board = esp32thing

;board = sparkfun_lora_gateway_1-channel

;board = ttgo-lora32-v1

;board = ttgo-t-beam

;board = turta_iot_node

;board = lolin_d32

;board = lolin_d32_pro

;board = lolin32

;board = wemosbat

;board = widora-air

;board = xinabox_cw02

;board = iotbusio

;board = iotbusproteus

;board = nina_w10

 

platform = espressif32

framework = arduino

lib_deps =

    https://github.com/dvarrel/AsyncTCP.git

    https://github.com/me-no-dev/ESPAsyncWebServer.git

thelastoutpostworkshop, 2024-01-06:
The problem you have is related to the ESPAsyncWebServer library, try examples from the ESPAsyncWebServer, they should compile and work in your environment if it is installed correctly.  After that, try the GPIOLibrary, example.

thelastoutpostworkshop, 2024-01-06:
@CharlesGodwin on what version of ESP32 boards are you ?  You should use the latest version 2.0.14

rescalle83, 2024-01-06:
I did try using the example for the ESPAsyncWebServer called simply server. This compiled fine. I went back and tried GPIOLibrary example and continue to get the following error

 

In file included from C:\Users\rober\OneDrive\Documents\Arduino\sketch_jan6a\sketch_jan6a.ino:7:

c:\Users\rober\OneDrive\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:11:10: fatal error: ESPAsyncWebServer.h: No such file or directory

   11 | #include <ESPAsyncWebServer.h>

      |          ^~~~~~~~~~~~~~~~~~~~~

compilation terminated.

 

exit status 1

 

Compilation error: exit status 1

CharlesGodwin, 2024-01-06:
> @CharlesGodwin on what version of ESP32 boards are you ?  You should use the latest version 2.0.14

I'll let you know Sunday

rescalle83, 2024-01-06:
The ESP32 version from Espressif is 3.0.0a. The one below is 2.0.11

knubbl1983, 2024-01-07:
Same problem here using a D1 Mini.

if u put defines like following in .h File the frist errors are gone.

/** Major version number (X.x.x) */
#define ESP_ARDUINO_VERSION_MAJOR   3
/** Minor version number (x.X.x) */
#define ESP_ARDUINO_VERSION_MINOR   0
/** Patch version number (x.x.X) */
#define ESP_ARDUINO_VERSION_PATCH   0

im Still struggling with the 

.pio\libdeps\wemos_d1_mini32\GPIOViewer\src/gpio_viewer.h:336:57: error: 'analogGetChannel' was not declared in this scope

failure

thelastoutpostworkshop, 2024-01-07:
> Same problem here using a D1 Mini.
>
> if u put defines like following in .h File the frist errors are gone.
>
> /** Major version number (X.x.x) _/ #define ESP_ARDUINO_VERSION_MAJOR 3 /_* Minor version [...]

Make sure you have the latest stable release of GPIOViewer.
Make sure you have the latest ESP32 boards definition from Espressif, the latest currentky are 2.0.14

knubbl1983, 2024-01-07:
> Make sure you have the latest stable release of GPIOViewer. Make sure you have the latest ESP32 boards definition from Espressif, the latest currentky are 2.0.14

i have latest release of GPIOViewer, since i have installed it today.
i alo think i have current version of espressif.

The quesiton is where are the definition for major minor and patch level in ur code?

thelastoutpostworkshop, 2024-01-07:
They are not in my code, they are defined in Espressif ESP32 Core code.
If you are using PlatformIO, go to Platforms in PIOHome, and update the Espressif32 boards.
If you are using Arduino IDE, go to the boards manager, and update to 2.0.14 for Espressif Systems

knubbl1983, 2024-01-07:
I have 3.5.0...is this correct?

[image]

thelastoutpostworkshop, 2024-01-07:
yes analogGetChannel() is in Espressif.
The latest Espressif board is :
[image]
Click on the update tab to update all the Espressif boards :
[image]

knubbl1983, 2024-01-07:
Tanks a lot . The old Espressif32 Version was the problem. Updated it to 6.5.0 everything is working.

Thread can be closed

thelastoutpostworkshop, 2024-01-07:
Glad it's working now

closed 2024-01-07 by thelastoutpostworkshop

rescalle83, 2024-01-07:
My issue still remains. It is not a library issue. I am (have to) use the board library from Esperrif as I have a C6 dev board. I have tried different machines, loaded in different folders, etc.. with no luck getting GPIOViewer to compile. I can work with ESPAsyncWebSvr with other examples but it will simply not compile with GPIOViewer.

 

I assume it has something to do with the 3.0.14a board version from Esperrif as the 2.0.014 does not support C6. The only boards I have are the ESP32-C6-DevKitC-1 (board version 1.2).

 

I have tried everything and I get the same error. The libraries are there, the folders are correct, the paths are good. Simply will not compile.

 

Hopefully someone can try this with the C6 in the Arduino environment using the 3.0.14a board package from Esperrif.

 

Appreciate all the help

Bob

rescalle83, 2024-01-07:
Did you update from Arduino IDE?

rescalle83, 2024-01-07:
Unfortunately 2.0.14 does not support the C6 iteration. They recommend to use the dev (not stable) version which is 3.0.14a and does support C6. I have even tried going to the 2.0.14 with still no luck on compile. Same error on 3 different machines using different folders and paths just to ensure something simple is caught.

 

Hopefully Espressif will move the C6 support into the stable board package at some point. Maybe there is something in the dev package that simply will not allow this to compile. All of my Blynk projects compile and work perfectly. I have compiled 9-10 different projects and all work with the dev package. I am sure I am missing something and will continue to try from time to time.

 

Really appreciate all the help and glad DKnapp was able to get things working!