thelastoutpostworkshop/gpio_viewer #60: Newbie here and getting the following errors when trying to compile

thelastoutpostworkshop, 2024-01-06:
Newbie here and getting the following errors when trying to compile
In file included from C:\Users\rober\AppData\Local\Temp\.arduinoIDE-unsaved202406-12528-1d9cd3.mz2qk\gpioviewer\gpioviewer.ino:34:
C:\Users\rober\OneDrive\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:11:10: fatal error: ESPAsyncWebServer.h: No such file or directory
   11 | #include <ESPAsyncWebServer.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1

_Originally posted by @rescalle83 in https://github.com/thelastoutpostworkshop/gpio_viewer/issues/58#issuecomment-1879755572_

thelastoutpostworkshop, 2024-01-06:
@rescalle83 did you include the  ESPAsyncWebServer ?

rescalle83, 2024-01-06:
Yes

thelastoutpostworkshop, 2024-01-06:
The library ESPAsyncWebServer is not installed, recheck.
Try some examples that comes with the ESPAsyncWebServer.

rescalle83, 2024-01-06:
I have installed and uninstalled several times but will try again. I can see the folder and files in libraries

rescalle83, 2024-01-06:
ESPAsyncWebServer works great from one of the examples like simple server. But when I try and use the GPIOViewer example, I get the error below. I have tried this on multiple machines and not sure what the issue is.

 

In file included from C:\Users\rober\AppData\Local\Temp\.arduinoIDE-unsaved202406-5396-y313rb.c4qme\gpioviewer\gpioviewer.ino:8:

c:\Users\rober\OneDrive\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:11:10: fatal error: ESPAsyncWebServer.h: No such file or directory

   11 | #include <ESPAsyncWebServer.h>

      |          ^~~~~~~~~~~~~~~~~~~~~

compilation terminated.

 

exit status 1

 

Compilation error: exit status 1

thelastoutpostworkshop, 2024-01-06:
@rescalle83 you may have problems in your Arduino IDE setup.  If your are getting this message : 

c:\Users\rober\OneDrive\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:11:10: fatal error: ESPAsyncWebServer.h: No such file or directory

It is because the ESPAsyncWebServer cannot be found in the project where you are using the GPIOViewer library.

rescalle83, 2024-01-06:
Not sure what different settings I can use. I can compile everything except GPIO. I even changed from DEV board to stable and loaded 2.0.14 and still have the same issue.

 

I can uninstall again and try to install it again. Not sure what I am doing wrong … odd

felipedsouza, 2024-01-07:
I have an error related to that:

Installed the FreeRTOS Library but didn't work

Arduino\libraries\GPIOViewer\src/gpio_viewer.h:17:10: fatal error: FreeRTOS.h: No such file or directory
   17 | #include <FreeRTOS.h>
      |          ^~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1

thelastoutpostworkshop, 2024-01-07:
See the troubleshooting section : https://github.com/thelastoutpostworkshop/gpio_viewer#troubleshooting

rescalle83, 2024-01-07:
Thanks. Followed this and no issues compiling without GPIOViewer. Once I add and try to compile, I get the same error. It may be I am using the C6 devkit. I have tried everything and the only variable that prevents compile is GPIOViewer. I will try again at some point when C6 is supported.

Thanks.

thelastoutpostworkshop, 2024-01-08:
Please add print screen of your boards version and boards selected, like this:
[image]
[image]

rescalle83, 2024-01-08:
image.png (view on web) <https://github.com/thelastoutpostworkshop/gpio_viewer/assets/127604345/917c17ea-d1ec-4981-bd77-1249a9fa401b> 
image.png (view on web) <https://github.com/thelastoutpostworkshop/gpio_viewer/assets/127604345/fff6f672-94e7-4068-afad-f0035d42eba2>

thelastoutpostworkshop, 2024-01-08:
I dot not see your print screens, maybe they do not show up when your answer by email.  Go to github and add them in this issue.

rescalle83, 2024-01-08:
[image]
[image]

thelastoutpostworkshop, 2024-01-08:
I have tested it only on stable release 2.011 - 2.014

rescalle83, 2024-01-09:
Thanks. Hopefully there will be support for C6 and the latest board package. I will continue to watch for future support.

 

The board I need support for is the ESP32-C6-DevKitC-1

closed 2024-01-11 by thelastoutpostworkshop