thelastoutpostworkshop/gpio_viewer #61: only Reboot why?

Errurion22, 2024-01-06:
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
GPIOViewer >> Connecting to WiFi...
...GPIOViewer >> Connected to WiFi
Attempting MQTT connection...
Connected to MQTT broker
GPIOViewer >> Release 1.0.7
GPIOViewer >> ESP32 Core Version 2.0.11
GPIOViewer >> Web Application URL is: http://192.168.178.65:8080
E (2769) ledc: ledc_get_dutyGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (GPIO Monitor Ta) 
Core  1 register dump:
PC      : 0x4008e2b1  PS      : 0x00060036  A0      : 0x4008c5cf  A1      : 0x3ffde4f0  
A2      : 0x3ffc5d08  A3      : 0x3ffc5ee8  A4      : 0x3ffc5eec  A5      : 0x00000001  
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x3ffc5d10  A9      : 0x3ffde4d0  
A10     : 0x3ffdecf8  A11     : 0x3ffdecf8  A12     : 0x00000014  A13     : 0x00000004  
A14     : 0x3ffdecf0  A15     : 0x80000001  SAR     : 0x0000001d  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4008a561  LEND    : 0x4008a571  LCOUNT  : 0xfffffffa  

Backtrace: 0x4008e2ae:0x3ffde4f0 0x4008c5cc:0x3ffde520 0x4008c57c:0x3ffde540 0x4008ce79:0x00000000 |<-CORRUPTED

thelastoutpostworkshop, 2024-01-06:
And if you comment out the line where GPIO calls begin, no reboot ?
What board are you using ?
Are you on PlatformIO ?

Errurion22, 2024-01-07:
My board is a az-delivery-devkit-v2
I use the Arduino IDE 2.2.1

I think I found the problem too...
I use the lib ESP32Servo.h and when I use myservo.attach(27); If I activate it, I get reboots

thelastoutpostworkshop, 2024-01-07:
If you found that the library is causing the problem, provide a simple sketch that reproduce it

Errurion22, 2024-01-08:
/***
This example is intended to demonstrate the use of the GPIO Viewer Library.

Tutorial : https://youtu.be/UxkOosaNohU
Documentation : https://github.com/thelastoutpostworkshop/gpio_viewer
***/

#include <ESP32Servo.h> // <---Integrating the lib works without any problems!!!!//

#include <gpio_viewer.h> // Must me the first include in your project
GPIOViewer gpio_viewer;
Servo myservo;

void setup() {
  delay(5000);
  Serial.begin(115200);

  myservo.attach(14); // <--- when this is activated it's over and I get reboots....

  // Comment the next line, If your code aleady include connection to Wifi in mode WIFI_STA (WIFI_AP and WIFI_AP_STA are not supported)
  gpio_viewer.connectToWifi("Nicht Therapierbar2", "W491G7op9mbKqhHhQv5F");
  // gpio_viewer.setPort(5555);                 // You can set the http port, if not set default port is 8080

  // Your own setup code start here

  // Must be at the end of your setup
  // gpio_viewer.setSamplingInterval(25); // You can set the sampling interval in ms, if not set default is 100ms

  gpio_viewer.begin();
}

// You don't need to change your loop function
void loop() {

}
// The rest of your code here

thelastoutpostworkshop, 2024-01-08:
I ran successfully the code you provided on an ESP32S3, I don't have your specific board.  Could you put a link to the board you are using  for az-delivery-devkit-v2 (on amazon for example).
Did you try your code without the GPIOViewer library, maybe you cannot use pin 14 on your specific board ?

Errurion22, 2024-01-09:
https://www.amazon.de/gp/product/B0BF5NRF8H/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&th=1
No matter which pin I use for the servo, it doesn't work
and yes my code works without the gpio_viewer

thelastoutpostworkshop, 2024-01-09:
Try with upgrading your esp32 boads at version 2.0.14, I see in your first message that your are using version 2.0.11

closed 2024-01-13 by thelastoutpostworkshop