thelastoutpostworkshop/gpio_viewer #65: script blocking on connection failure

nicolas-j, 2024-01-07:
In the event of a connection failure, or in the absence of wifi access, scripts using gpioviewer remain blocked on the connection attempt. Would it be possible to define a time delay that would allow the script to continue if the connection fails?

thelastoutpostworkshop, 2024-01-07:
GPIOViewer cannot work without a Wifi connection.

nicolas-j, 2024-01-07:
I understand. I thought it would be usefull in case of testing at home with wifi, then using the same code away from wifi without any modification of my script. gpioviewer could return some bolean value in case of failed connection my main script could use.

thelastoutpostworkshop, 2024-01-07:
give me an code example

nicolas-j, 2024-01-07:
I am not that good at coding, but if I look at something I used before, the modification could be inspired by this method:

public:
    // ... 
    bool begin() {
        Serial.printf("GPIOViewer >> Release %s\n", release);
        Serial.printf("GPIOViewer >> ESP32 Core Version %d.%d.%d\n", ESP_ARDUINO_VERSION_MAJOR, ESP_ARDUINO_VERSION_MINOR, ESP_ARDUINO_VERSION_PATCH);
        
        if (ESP_ARDUINO_VERSION_MAJOR < 2) {
            Serial.printf("GPIOViewer >> Your ESP32 Core Version etc...;
            return false;
        }
        if (checkWifiStatus()) {
            // ...configurations, initializations
            return true; // if the Wi-Fi connection is successfully established
        } 
        return false; // if the Wi-Fi connection fails
    }
};

Then, in my main code, it would be possible to check the value returned:

void setup() {
 
    GPIOViewer gpioViewer;
    gpioViewer.connectToWifi("YourSSID", "YourPassword");
    if (gpioViewer.begin()) {
        // success, go on to the usefull code
    } else {
        // fail, code to handle it
    }
}

End of my abilities.

thelastoutpostworkshop, 2024-01-07:
Now I understand what you want to achieve.  The call gpioViewer.begin() must be **the very last line in your setup**, if it's not, some pins could failed to be monitored. So nothing else should be called after it.
What do you want to handle exactly if GPIOViewer.begin() fails ?

nicolas-j, 2024-01-07:
I would like my loop to start even if connection fails.
My code lights up leds depending of some sensors inputs. the loop is checking these sensors at regular intervals. Adding gpioviewer stops the execution of the loop when wifi connection fails.

nicolas-j, 2024-01-07:
I could also use this function to set pin 14 to "low", since wifi wouldn't be necessary to me in the event of a connection failure.

thelastoutpostworkshop, 2024-01-07:
Do you get any Serial Monitor messages when Wifi fails ?
Because GPIOViewer is using WiFi.begin function from the wifi library, so if it fails, your loop function should execute, unless connection to Wifi is trying to be established indefinitely by the wifi library.  This happens when I enter a wrong ssid and paswword.

nicolas-j, 2024-01-07:
When I turn my router wifi off, the serial monitor returns:
16:05:49.684 -> GPIOViewer >> Connecting to WiFi...
16:05:50.165 -> ......................................................................................................
and keeps on searching, the loop does not start

thelastoutpostworkshop, 2024-01-07:
yes that's the default behavior of the Wifi library, I do not have control over that library, it's part of the Arduino Framework

thelastoutpostworkshop, 2024-01-07:
I could also add a timeout.  Meanwhile you could comment out gpio_viewer.connectToWifi(ssid, password); and do your own wifi connection in the setup with a timeout.

closed 2024-01-08 by thelastoutpostworkshop