thelastoutpostworkshop/gpio_viewer #68: Lack of ESP8266 board support

arendst, 2024-01-07:
I've refactored your gpio_viewer.h to Tasmota. The current Tasmota dev branch supports both it when enabled with define USE_GPIO_VIEWER or optionally using a berry app.

I refactored in such a way to get rid of both outdated AsyncTCP and ESPAsyncWebServer libraries and use the default Arduino Core WebServer. This also prepares for ESP32 Arduino core 3.0

As it now runs on ESP8266 too it would be nice to have some ESP8266 boards like (at least) generic and a wemos D1.

Pls consider to add ESP8266 boards to your repository.

Mergenth, 2024-02-26:
pls click the **like** = heart above  and **subscribe** for this thread.

Mergenth, 2024-05-20:
Pls

Mergenth, 2024-06-20:
Pls provide that as a next - please.

homeplayer2, 2024-07-03:
> As it now runs on ESP8266 too

I am not new to programming or esp Arduino electronics I am new to github
I have tried;
Pop os vsc and codium, flatpack and native install Arduino ide  and even on a debian system to get Gpio Viewer working on ESP8266 Have this problem [https://github.com/thelastoutpostworkshop/gpio_viewer/issues/111](url)

sonukmrsah, 2025-07-20:
I started working on this issue, and I got to know that both ESP32 and ESP8266 have different architectures. My first thought was to check for ESP8266-specific functionality, and with the help of the header guard, I can have different code executions for different architectures. 

It made the code bulkier and harder to understand. I am looking for any resource or help designing this functionality.

closed 2025-11-29 by thelastoutpostworkshop