thelastoutpostworkshop/gpio_viewer #73: Feature Request: Ability to reverse the webpage indicator colours for LOW and HIGH states

toetoast, 2024-01-07:
With regards to the webpage indicator colours for LOW & HIGH states, there appears to be no universal standard, but rather a common choice made by many manufacturers for consistency and ease of use.

I'd like to suggest please a new Library Method that allows the webpage GPIO indicator colours to be set within code so that:

1. HIGH=Red & LOW=Green (the current mode of operation)
or 
2. HIGH=Green & LOW=Red

Thanks
Tony

thelastoutpostworkshop, 2024-01-07:
Good Idea!

thelastoutpostworkshop, 2024-01-07:
Added this indicator when the WebApplication receives update from pins.  When there are no pins activities for 1 second, the GPIOViewe library will resend to show some activity (next release)

toetoast, 2024-01-08:
Sounds great, thank you.
This GPIO Viewer is an amazing tool, which is getting better and better since your YouTube video was released 10 days ago!

closed 2024-01-08 by thelastoutpostworkshop