thelastoutpostworkshop/gpio_viewer #74: Compiling Error when compiling the example

jontubs, 2024-01-07:
Hello,

I installed all libraries and board descriptions like in your youtube video

But when I try to compile (even without uploading) this error

In file included from C:\Users\xxx\AppData\Local\Temp\.arduinoIDE-unsaved202407-9676-1wk0q39.n6b7j\gpioviewer\gpioviewer.ino:8:
c:\Users\xxx\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h: In member function 'int GPIOViewer::readGPIO(int, uint32_t*, pinTypes*)':
c:\Users\xxx\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:346:33: error: 'analogGetChannel' was not declared in this scope; did you mean 'analogChannel'?
  346 |         uint8_t analogChannel = analogGetChannel(gpioNum);
      |                                 ^~~~~~~~~~~~~~~~
      |                                 analogChannel

exit status 1

Compilation error: exit status 1

I tried it on multiple PCs and also in a clean installed virtual machine.

What I am doing wrong?

thelastoutpostworkshop, 2024-01-07:
what board are you using ?
what is your IDE ?

jontubs, 2024-01-08:
I am using the newest version of the Arduino IDE 
Version: 2.2.1
Date: 2023-08-31T14:35:44.802Z
CLI Version: 0.34.0

I have to kinds of board here. An original lolin s2 mini and a XIAO-ESP32-C3. But it doesnt matter which board I am choosing from the device list. The error already accures during compiling without any board connected.

Is there another setting to select the board except the "Board Manager"?
[image]

jontubs, 2024-01-08:
I am using the Arduino IDE

jontubs, 2024-01-08:
I am using the original Arduino IDE

Version: 2.2.1
Date: 2023-08-31T14:35:44.802Z
CLI Version: 0.34.0

Copyright © 2024 Arduino SA

I have two kinds of board here: an original Lolin S2 Mini and Nano ESP32 

Is there another setting to select the Board then the "Board Manager"
[image]

thelastoutpostworkshop, 2024-01-08:
you should also have the latest version of ESP32 Code, look in your board manager, you should see this, update if it's not 2.0.14:
[image]

jontubs, 2024-01-08:
okay, there is difference. I have an even newer version

[image]

When I Force-Install the 2.0.14 Version it compiles.

Thank you. Should this be considered in the software?

thelastoutpostworkshop, 2024-01-08:
yes I will add the check for the alpha version

closed 2024-01-09 by thelastoutpostworkshop