thelastoutpostworkshop/gpio_viewer #76: not compiled for XIAO C3

ErfanDL, 2024-01-08:
Hi. I got this error when compiling for XIAO C3

c:\Users\STRIX-PC\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\AsyncWebSocket.cpp: In member function 'IPAddress AsyncWebSocketClient::remoteIP()':
c:\Users\STRIX-PC\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\AsyncWebSocket.cpp:832:28: error: call of overloaded 'IPAddress(unsigned int)' is ambiguous
         return IPAddress(0U);
                            ^
In file included from C:\Users\STRIX-PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/Arduino.h:180,
                 from c:\Users\STRIX-PC\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\AsyncWebSocket.cpp:21:
C:\Users\STRIX-PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/IPAddress.h:51:5: note: candidate: 'IPAddress::IPAddress(const uint8_t*)'
     IPAddress(const uint8_t *address);
     ^~~~~~~~~
C:\Users\STRIX-PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/IPAddress.h:50:5: note: candidate: 'IPAddress::IPAddress(uint32_t)'
     IPAddress(uint32_t address);
     ^~~~~~~~~
C:\Users\STRIX-PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/IPAddress.h:29:7: note: candidate: 'constexpr IPAddress::IPAddress(const IPAddress&)'
 class IPAddress: public Printable
       ^~~~~~~~~

exit status 1

Compilation error: exit status 1

thelastoutpostworkshop, 2024-01-08:
You seams to have forget to install the AyncTCP Library

ErfanDL, 2024-01-09:
> You seams to have forget to install the AyncTCP Library

but required libraries are installed !

thelastoutpostworkshop, 2024-01-09:
something is wrong in your libraries installation

closed 2024-01-13 by thelastoutpostworkshop