thelastoutpostworkshop/gpio_viewer #77: NodeMCU ESp8266

paulsjvr, 2024-01-08:
Hi
Please Add NodeMCU ESp8266 to GPIO viewer.

Thanx for the great work.

Paul

thelastoutpostworkshop, 2024-01-08:
I am not sure if ESP8266 will be compatible, I will do some testing

arendst, 2024-01-08:
Try Tasmota dev branch. Already fitted with a functional adaptation of gpio_viewer and tested on ESP8266 and all current ESP32 derivatives like ESP32-C3 but also -C2 and -C6 using Arduino Core 3.0

[image]

The above picture is from an wemos Di mini (ESP8266) running Tasmota v13.30.2

See here for adaptations (mainly removing some ESP32 specifics) https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_xdrv_driver/xdrv_121_gpioviewer.ino

paulsjvr, 2024-01-09:
> Try Tasmota dev branch. Already fitted with a functional adaptation of gpio_viewer and tested on ESP8266 and all current ESP32 derivatives like ESP32-C3 but also -C2 and -C6 using Arduino Core 3.0 [...]

Awesome
Thanx Theo, I will give it a try.

closed 2024-01-13 by thelastoutpostworkshop