thelastoutpostworkshop/gpio_viewer #78: Firebeetle esp32

KennyM64, 2024-01-09:
Hi, I like the concept but it doesn't seem to work with boards that have dedicated Analog ports. I'm using a Firebeetle ESP32 for IoT as it has 5 dedicated analog ports 15, 34, 35, 36, and 39. However only 15 works in a simple sketch with the others throwing a 
"E (282666) ledc: ledc_channel_config(608): gpio_num argument is invalid" error.

Anyway around this?

https://wiki.dfrobot.com/FireBeetle_ESP32_IOT_Microcontroller(V3.0)__Supports_Wi-Fi_&_Bluetooth__SKU__DFR0478
  analogWrite(15, 256);
  delay(1000);
  analogWrite(35, 512);
  delay(1000);
  analogWrite(34, 1024);
    delay(1000);                      // wait for a second
  analogWrite(39, 256);
  delay(1000);
  analogWrite(36, 512);
  delay(1000);
  analogWrite(15, 1024);

thelastoutpostworkshop, 2024-01-09:
if you remove the GPIOViewer Library and run the code above, do you get the error message ?
If you go into the analogWrite function, can you print here the code you get for that function

KennyM64, 2024-01-10:
Apologies. It seems to be an issue with the board and how it's assigning
dedicated Analog ports to the Firebeetle ESP32. A0-A4 use
GPIO's 36,39,34,35,15. If I change the board to an ESP32 Dev Kit I get the
same errors. Looking at the pinout
<https://mischianti.org/wp-content/uploads/2020/11/ESP32-DOIT-DEV-KIT-v1-pinout-mischianti.png>for
the  ESP32 Dev Kit it shows 36,39,34,35 as input only. So I presume this
carried over to the Firebeetle board but isn't documented
<https://raw.githubusercontent.com/Frances9/DFR0478/master/DFR0478%E5%BC%95%E8%84%9A%E5%9B%BE01.png>as
such.

All good :-) Well done on the code it will be very helpful. I'll keep an
eye out for new boards.

thelastoutpostworkshop, 2024-01-10:
Thanks, glad it is not working.

closed 2024-01-10 by thelastoutpostworkshop