thelastoutpostworkshop/gpio_viewer #79: 'analogGetChannel' was not declared in this scope

samarkh, 2024-01-09:
This could be the same as issue #74 
When I try the basic Platformio example I get
 'analogGetChannel' was not declared in this scope
 gpio_viewer.h, line 374
platformio.ini

```
[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit

framework = arduino
monitor_speed = 115200
upload_speed = 1500000
monitor_port = COM11
upload_port = COM11
lib_deps =
            thelastoutpostworkshop/GPIOViewer @ ^1.0.7
            https://github.com/me-no-dev/ESPAsyncWebServer.git
            https://github.com/dvarrel/AsyncTCP.git
            https://github.com/me-no-dev/ESPAsyncWebServer.git
build_flags = 
  -D CORE_DEBUG_LEVEL=5

```
Any sugestions?
Yours Simon M

thelastoutpostworkshop, 2024-01-09:
You need tp update your platform espressif32 to the latest version (2.0.14)

toetoast, 2024-01-09:
If for some reason you need the installed Espressif Systems ESP32 Board (accessed via Arduino Boards Manager) to be less then 2.0.14 then the lowest you can go for compatibility is 2.0.5 as shown here.

The documentation within the README file has now been updated to reflect this.

Tony

[image]

samarkh, 2024-01-10:
Many thanks
`PlatformIO Core CLI → pio pkg update -g -p espressif32 `
worked a charm

closed 2024-01-10 by samarkh