thelastoutpostworkshop/gpio_viewer #81: [Idea] Consider drop down menu based on chip name

arendst, 2024-01-13:
With ever growing amount of supported dev boards the drop down list becomes a little unusable. Consider adding a ``chip`` field pair to the boards.json file to selectively show chip based drop down list.

A starting point could be the used names in esptool (and Tasmota ;-). For version 4.7.0 I extracted the below list:
```
chip_name = {
    0: "ESP8266EX"        
    1: "ESP8285H08" if max_temp else "ESP8285N08",
    2: "ESP8285H16" if max_temp else "ESP8285N16",
}
Test pattern: ESP8266 and ESP8285

        
chip_name = {
    0: "ESP32-C2",
    1: "ESP32-C2",
}        
Test pattern: ESP32-C2

        
chip_name = {
    0: "ESP32-C3 (QFN32)",
    1: "ESP8685 (QFN28)",
    2: "ESP32-C3 AZ (QFN32)",
    3: "ESP8686 (QFN24)",
}        
Test pattern: ESP32-C3 and ESP8685 and ESP8686

        
chip_name = {
    0: "ESP32-C6 (QFN40)",
    1: "ESP32-C6FH4 (QFN32)",
}        
Test pattern: ESP32-C6

        
chip_name = {
    0: "ESP32-H2",
}        
Test pattern: ESP32-H2

        
chip_name = {
    0: "ESP32-S2",
    1: "ESP32-S2FH2",
    2: "ESP32-S2FH4",
    102: "ESP32-S2FNR2",
    100: "ESP32-S2R2",
}        
Test pattern: ESP32-S2

        
chip_name = {
    0: "ESP32-S3 (QFN56)",
    1: "ESP32-S3-PICO-1 (LGA56)",
}        
Test pattern: ESP32-S3

        
chip_name = {
    0: "ESP32-S0WDQ6" if single_core else "ESP32-D0WDQ6",
    1: "ESP32-S0WD" if single_core else "ESP32-D0WD",
    2: "ESP32-D2WD",
    4: "ESP32-U4WDH",
    5: "ESP32-PICO-V3" if rev3 else "ESP32-PICO-D4",
    6: "ESP32-PICO-V3-02",
    7: "ESP32-D0WDR2-V3",
}
Test pattern: If all else fails use ESP32
```

Going over the above list from top to bottom allows to fall back to all ESP32 boards.

thelastoutpostworkshop, 2024-01-13:
I am willing to use standard names, where can I get them ?

arendst, 2024-01-13:
The "chip_name" names as shown before are from the espressif tool ``esptool``. 

As you can see there are many different types of chips even for a simple esp32.

I suggest to use the test pattern names as "standard" per chip type.

thelastoutpostworkshop, 2024-01-13:
That makes sense, I wonder if it is easy to find the chip used in all the boards already listed

arendst, 2024-01-13:
You'll have to make some assumptions. Users will make issues if they find the chip is wrong. 

From what I can see most of them are "ESP32". If you see a C3 in the name it's "ESP32-C3", S2 = "ESP32-S2", S3 = "ESP32-S3", 8266 = "ESP8266"

thelastoutpostworkshop, 2024-01-13:
I am considering adding the options for the users to see boards in the menu by chip names or by board names (sorted and in sub level menus).
I will do this in the next major release, using vuejs and a UI Framework (like Vuetify) to make the app with a modern UI and avoid fighting with screen sizing.

toetoast, 2024-01-13:
Just to help out here, I can tell you that the TinyPICO V3 and TinyPICO Nano boards both use the ESP32 PICO-D4 chip.

toetoast, 2024-01-13:
One other additional suggestion here might be to make the drop down list Font size about half the size that it currently is, so that more boards can fit on the screen when the list is shown.