thelastoutpostworkshop/gpio_viewer #92: Wrong schematic

RobDel99, 2024-02-02:
Wow, amazing work you’ve done with this library. Thanks a zillions! 

In the board posting there’s few board schematic that don’t match. 

Freenove ESP32-S3, wrong schematic
Lilygo T7 Mini32 v1.5, Only half the picture show up
XIAO-ESP32-C3, has the Freenove ESP32-S3 schematic
XIAO-ESP32-C3, is repeated at the last of the list. 
Wemos Lolin S3 Mini, the schematic can be found here: [image]

one additional board will be the XIAO-Seeeduino ESP32-S3 
 - Schematic : https://github.com/thelastoutpostworkshop/gpio_viewer/assets/122332111/d2969c39-0ee6-4dd6-bc77-4d4161f35fa4
 - Picture : https://github.com/thelastoutpostworkshop/gpio_viewer/assets/122332111/2f05c979-3344-4e46-8d3c-129d931e3d7b

Thanks and take care,

Robert

thelastoutpostworkshop, 2024-02-02:
thanks I have corrected the schematics

thelastoutpostworkshop, 2024-02-02:
The board XIAO-Seeeduino ESP32-S3 is now available

closed 2024-02-02 by thelastoutpostworkshop