thelastoutpostworkshop/gpio_viewer #95: Is the "Pins" button on version 1.5 Web UI doing the opposite operation?

toetoast, 2024-02-09:
Hi,

I've been using version 1.5 for a little while now and I'm wondering if the "Pins" button is doing the opposite to what I think it should be doing?

When pressed and it goes blue, the ESP32 pin numbers are turned off. Should this do the opposite?

Thanks
Tony

thelastoutpostworkshop, 2024-02-09:
It depends if the image of the board have pin numbers, in that case the pin types are shown,  I will change the UI to be more accurate on what it does.

toetoast, 2024-02-09:
Thanks. I've been using the following boards, but I'm guessing this happens to all boards:

- Generic View
- TinyPICO V3
- TinyPICO Nano

thelastoutpostworkshop, 2024-02-09:
I have updated the app (v2.0.8) now the pins button should be more obvious,
Let me know what you think.

toetoast, 2024-02-10:
That works much better and it easy to understand, since the button now changes from "Pin Types" to "Pin Numbers".
So its very much now like a GPIO pin "Mode" button.

Maybe the icon no longer needs to change to blue then grey each time its pressed? This is because the button now shows the text each of the 2 modes.

thelastoutpostworkshop, 2024-02-15:
The icon color for the pin button on the bottom bar has been removed in the new release v2.0.9 of the web application.

closed 2024-02-15 by thelastoutpostworkshop