thelastoutpostworkshop/gpio_viewer #103: GPIO Viewer

Alzburg, 2024-02-24:
In file included from C:\Users\jhmax\AppData\Local\Temp\.arduinoIDE-unsaved2024124-5872-1df5n7z.rb36\gpioviewer\gpioviewer.ino:8:
c:\Scatchbook\libraries\GPIOViewer\src/gpio_viewer.h: In member function 'int GPIOViewer::readGPIO(int, uint32_t*, pinTypes*)':
c:\Scatchbook\libraries\GPIOViewer\src/gpio_viewer.h:525:33: error: 'analogGetChannel' was not declared in this scope; did you mean 'analogChannel'?
  525 |         uint8_t analogChannel = analogGetChannel(gpioNum);
      |                                 ^~~~~~~~~~~~~~~~
      |                                 analogChannel

exit status 1

Compilation error: exit status 1

Hi everyone, I'm trying to install the GPIO Viewer on the ESP32 Dev Module and it keeps giving me this error, can you please advise me what I'm doing wrong when I did everything according to the instructions and the Youtube video? Thank you and have a nice day.

thelastoutpostworkshop, 2024-02-24:
Your probably not on the latest version of ESP32 boards for Arduino.  Update to the latest version.

Alzburg, 2024-02-24:
In file included from C:\Scatchbook\gpioviewer\gpioviewer.ino:8:
c:\Scatchbook\libraries\GPIOViewer\src/gpio_viewer.h: In member function 'int GPIOViewer::readGPIO(int, uint32_t*, pinTypes*)':
c:\Scatchbook\libraries\GPIOViewer\src/gpio_viewer.h:525:33: error: 'analogGetChannel' was not declared in this scope; did you mean 'analogChannel'?
  525 |         uint8_t analogChannel = analogGetChannel(gpioNum);
      |                                 ^~~~~~~~~~~~~~~~
      |                                 analogChannel

exit status 1

Compilation error: exit status 1

https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

I tried the ESP update from this page, then restarted the Arduino IDE and it says this. I'm already an old man of 60 years, so I can't move about it somehow :-( Thank you for your effort!

thelastoutpostworkshop, 2024-02-24:
You have to do the board update through the Arduino IDE in the boars manager.

Alzburg, 2024-02-24:
Do you have a name please, besides the ones I have installed?
[image]

Alzburg, 2024-02-24:
[image]

Alzburg, 2024-02-24:
Do you have a name please, besides the ones I have installed?

thelastoutpostworkshop, 2024-02-24:
You have an alpha version installed (3.0.0a) in the dropdown menu, choose version 2.0.14 
[image]

Alzburg, 2024-02-24:
[image]

Alzburg, 2024-02-24:
I don't have this version.

thelastoutpostworkshop, 2024-02-24:
2.0.11 should work

Alzburg, 2024-02-24:
Bingo! It works! Thank you very much for your quick help and have a nice night! Greetings from the Czech Republic! Hello!

thelastoutpostworkshop, 2024-02-24:
Glad it works!

closed 2024-02-24 by thelastoutpostworkshop