thelastoutpostworkshop/gpio_viewer #106: GPIOViewer compile error

NLoebel, 2024-03-03:
Hmm.  Did anything change with Arduino IDE 2.3.2 maybe?  Using ESP32-S2 with 2MB PSRAM.  Compiler says:

In member function 'void GPIOViewer::begin()':
error: format '%s' expects argument of type 'char*', but argument 3 has type 'String' [-Werror=format=]
             
Serial.printf("GPIOViewer >> PSRAM Size %s\n", **formatBytes(psramSize)**);

thelastoutpostworkshop, 2024-03-03:
Mine compiled correctly, but it may depend on compiler configuration.
There is indeed a problem with that line, it should be: 
Serial.printf("GPIOViewer >> PSRAM Size %s\n", formatBytes(psramSize).c_str());

You can comment it out, until it is fixed in the next release.

NGL3324, 2024-03-03:
whoa!!  talk about cycle time on that response!! Awesome!!  thank you!!

I actually had already done that, and GPIOViewer then compiled & ran fine, & I was coming back here to update my comment only to see your kind response.  My code ran fine, spat out all the correct size parameters (I just put your PSRAM report into my own code without  formatters).  However - unfortunately - despite compiling fine & running - connection to the provided URL just times out.  Two different browsers.  Man I was looking forward to seeing that cool ESP32 live graphic!!

NGL3324, 2024-03-03:
WORKED ON MOBILE! It's an issue with Xfinity router permissions (on any port) I'd guess.  Who cares, working on mobile's good enough!  Thanks for your awesome app!

thelastoutpostworkshop, 2024-03-04:
This has been fixed in release 1.5.1

closed 2024-03-04 by thelastoutpostworkshop