thelastoutpostworkshop/gpio_viewer #108: Add Waveshare ESP32C3-Zero to GPIO-Viewer

thomas-lentz, 2024-03-30:
Can you please add the [Waveshare ESP32C3-Zero](https://www.waveshare.com/wiki/ESP32-C3-Zero) to GPIO-Vierwer?
[image]

Thank you for your work
Greetings
 Tom

thelastoutpostworkshop, 2024-03-30:
The board is now available.

closed 2024-03-30 by thelastoutpostworkshop

thomas-lentz, 2024-04-06:
Thank you, that was really quick. Now I have to figure out, why the model is not shown in the Tasmota GPIO-Viewer section...

arendst, 2024-04-06:
If you use Tasmota v13.3.0 you won't see it as this version uses a different hosted version of gpio_viewer for legacy reasons. Just change to the current live hosted gpio_viewer using command gpurl or install current tasmota dev release.