thelastoutpostworkshop/gpio_viewer #111: Problem when compiling in nodemce ESP8266 ( WebServer.h )

Nasseryui, 2024-04-12:
Hello,
could you help me please with this, I get an error while compiling the code in nodemce ESP8266, when I change the board everything is okey but when I chose the nodemce ESP8266 it show's this error, it's in the gpio_viewer.h
```
In file included from C:\Users\HP 17\Documents\Arduino\gpio.ino:23:
c:\Users\HP 17\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:5:10: fatal error: WebServer.h: No such file or directory
    5 | #include <WebServer.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
exit status 1
Compilation error: exit status 1
```

thelastoutpostworkshop, 2024-04-12:
ESP8266 is not supported yet, only ESP32 boards

thelastoutpostworkshop, 2024-04-12:
Can you provide a link to your board, I will buy one to make the library compatible

AlexC-0806, 2024-04-12:
I bought my ESP8266 NodeMCU board on Wish.com. here's the link to my board: https://dl.wish.com/3iBSs
I'd really love if you made this excellent library compatible with ESP8266 boards

Nasseryui, 2024-04-13:
I thought it's supported because I saw it in the ESP32 Boards Supported part.
> Can you provide a link to your board, I will buy one to make the library compatible

yes sure, you can find it in Aliexpress, this one [](https://fr.aliexpress.com/item/1005005077804800.html)

homeplayer2, 2024-07-05:
> Hello, could you help me please with this, I get an error while compiling the code in nodemcu ESP8266, when I change the board everything is okey but when I chose the nodemcu ESP8266 it show's this error, it's in the gpio_viewer.h
> 
> ```
> In file included from C:\Users\HP 17\Documents\Arduino\gpio.ino:23:
> c:\Users\HP 17\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:5:10: fatal error: WebServer.h: No such file or directory
>     5 | #include <WebServer.h>
>       |          ^~~~~~~~~~~~~
> compilation terminated.
> exit status 1
> Compilation error: exit status 1
> ```

I am not to new to programming ore ide have tryed to get it working on pop-os debian, flatpak install of vsc and codium even arduino ide tryed also native installs but same problem
 >     5 | #include <WebServer.h>
>       |          ^~~~~~~~~~~~~

doitranch, 2024-07-17:
Ditto on a ESP8266EX. Same error tried following the depends of ESPAsyncTCP but got nothing useful.

thelastoutpostworkshop, 2025-11-29:
Sorry but the ESP8266 is not supported

closed 2025-11-29 by thelastoutpostworkshop