thelastoutpostworkshop/gpio_viewer #116: ESP32 C3 Super Mini Dev Board

ScorchFeed, 2024-05-06:
Please Help. When I try to compile, I get this error:

In file included from c:\Users\mpken\OneDrive\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:11,
                 from C:\Users\mpken\AppData\Local\Temp\.arduinoIDE-unsaved202446-7204-2hb6t6.q0vwo\gpioviewer\gpioviewer.ino:9:
c:\Users\mpken\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:35:10: fatal error: AsyncTCP.h: No such file or directory
   35 | #include <AsyncTCP.h>
      |          ^~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

**_This is the Code I used:_**

/***
This example is intended to demonstrate the use of the GPIO Viewer Library.

Tutorial : https://youtu.be/UxkOosaNohU
Latest Features : https://youtu.be/JJzRXcQrl3I
Documentation : https://github.com/thelastoutpostworkshop/gpio_viewer
***/

#include <gpio_viewer.h> // Must me the first include in your project
GPIOViewer gpio_viewer;

void setup()
{
  Serial.begin(115200);

  // Comment the next line, If your code aleady include connection to Wifi in mode WIFI_STA (WIFI_AP and WIFI_AP_STA are not supported)
  gpio_viewer.connectToWifi("BelongAEDA8D", "enfwewwk4cxj");
  gpio_viewer.setPort(5555);                 // You can set the http port, if not set default port is 8080

  // Your own setup code start here

  // Must be at the end of your setup
  // gpio_viewer.setSamplingInterval(25); // You can set the sampling interval in ms, if not set default is 100ms
  gpio_viewer.begin();
}

// You don't need to change your loop function
void loop() {

}
// The rest of your code here

thelastoutpostworkshop, 2024-05-06:
The AsyncTCP.h library is not instllaed, you have to install it throught the Arduino IDE library manager.  See the installation instructions here: https://github.com/thelastoutpostworkshop/gpio_viewer

ScorchFeed, 2024-05-06:
More problems...Please help.

In file included from C:\Users\mpken\AppData\Local\Temp\.arduinoIDE-unsaved202446-8116-1yj938d.vwo9\sketch_may6a\sketch_may6a.ino:1:
c:\Users\mpken\OneDrive\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h: In member function 'int GPIOViewer::readGPIO(int, uint32_t*, pinTypes*)':
c:\Users\mpken\OneDrive\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:500:33: error: 'analogGetChannel' was not declared in this scope; did you mean 'analogChannel'?
  500 |         uint8_t analogChannel = analogGetChannel(gpioNum);
      |                                 ^~~~~~~~~~~~~~~~
      |                                 analogChannel
Multiple libraries were found for "ESPAsyncWebServer.h"
  Used: C:\Users\mpken\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer
  Not used: C:\Users\mpken\OneDrive\Documents\Arduino\libraries\ESP_Async_WebServer
exit status 1

Compilation error: exit status 1

ScorchFeed, 2024-05-06:
I removed the multiples.  This is the latest error message:............................

In file included from C:\Users\mpken\AppData\Local\Temp\.arduinoIDE-unsaved202446-8116-1yj938d.vwo9\sketch_may6a\sketch_may6a.ino:1:
c:\Users\mpken\OneDrive\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h: In member function 'int GPIOViewer::readGPIO(int, uint32_t*, pinTypes*)':
c:\Users\mpken\OneDrive\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:500:33: error: 'analogGetChannel' was not declared in this scope; did you mean 'analogChannel'?
  500 |         uint8_t analogChannel = analogGetChannel(gpioNum);
      |                                 ^~~~~~~~~~~~~~~~
      |                                 analogChannel

exit status 1

Compilation error: exit status 1

thelastoutpostworkshop, 2024-05-06:
In the Arduino IDE Boards Manager, make sure you have the latest version of the Espressif ESP32 boards :
[image]

ScorchFeed, 2024-05-07:
This is what I have installed.
[image]

thelastoutpostworkshop, 2024-05-07:
You are using the alpha version of the boards, this is not yet supported, use 2.0.14 version

thelastoutpostworkshop, 2024-05-07:
Glad it works!

closed 2024-05-07 by thelastoutpostworkshop

talbengal, 2025-03-24:
Getting this error:The system cannot find the path specified.
exit status 1

Compilation error: exit status 1

Please help