thelastoutpostworkshop/gpio_viewer #119: Request for an ESP32-devkitC board

enmann59, 2024-05-10:
GPIOviewer is a great tool!  I really enjoy it!  I'd appreciate support for the ESP32-devkitC board if/when you get an opportunity.

Links for information from the Espressif are below:
Board Picture: [https://www.espressif.com/sites/default/files/dev-board/ESP32-DevKitC_S_0.png](url)
Pin Description: [https://docs.espressif.com/projects/esp-idf/en/latest/esp32/_images/esp32-devkitC-v4-pinout.png](url)
ESP32-DevKitC getting started guide: [https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-devkitc.html](url)

For future reference, do you prefer links to board information or to have files attached?

Thank you for this terrific application!

thelastoutpostworkshop, 2024-05-10:
Thank you.  The way you submitted it is perfect, links with a board image.

enmann59, 2024-05-10:
Thank you!  This appear to be the same module as the ESP32 Wroom 32UE ... not sure if I missed this earlier (I thought I carefully checked all the modules) but it's here now!

Using ESP32 Wroom 32UE is working great for me

I will close this request with this comment.

BTW - went through your source code at the base URL (https://thelastoutpostworkshop.github.io/microcontroller_devkit/gpio_viewer_1_5) ... brilliant!  Thank you again.

enmann59, 2024-05-10:
I am using ESP32 Wroom 32UE now which appears to match the ESP32-DevKitC and it works great!  Thanks again for your excellent project!

closed 2024-05-10 by enmann59