thelastoutpostworkshop/gpio_viewer #120: Compile error

BenthamSoftwareServices, 2024-05-12:
I get the following when trying to compile the gpioviewer example on Arduino IDE 2.3.2:

In file included from C:\Users\Steven\AppData\Local\Temp\.arduinoIDE-unsaved2024412-25320-k4kcgk.f6mv\gpioviewer\gpioviewer.ino:8:
e:\Users\Steven\Documents\GitHub\Arduino_IDE2\Arduino_IDE2_Sketches\libraries\GPIOViewer\src/gpio_viewer.h: In member function 'int GPIOViewer::readGPIO(int, uint32_t*, pinTypes*)':
e:\Users\Steven\Documents\GitHub\Arduino_IDE2\Arduino_IDE2_Sketches\libraries\GPIOViewer\src/gpio_viewer.h:500:33: error: 'analogGetChannel' was not declared in this scope; did you mean 'analogChannel'?
  500 |         uint8_t analogChannel = analogGetChannel(gpioNum);
      |                                 ^~~~~~~~~~~~~~~~
      |                                 analogChannel

exit status 1

Compilation error: exit status 1

thelastoutpostworkshop, 2024-05-12:
See here : https://github.com/thelastoutpostworkshop/gpio_viewer/issues/116#issuecomment-2096921380

BenthamSoftwareServices, 2024-05-12:
That fixed it, many thanks.  Really useful app.

closed 2024-05-12 by thelastoutpostworkshop