thelastoutpostworkshop/gpio_viewer #121: Error when connecting to WiFi - ESP32-WROOM-32 resets constantly.

jramongit, 2024-05-16:
When using GPIO_Viewer, my ESP32-WROOM-32 is constantly rerebooting/reseting, restarting once and once again.

No matter if I use GPIO suggested line to connect to WiFi:

     gpio_viewer.connectToWifi (WIFI_SSID_1, WIFI_PASSWORD_1);   //Connect to WiFi

or the way I usually connect:

     #include <WiFiMulti.h>
     WiFiMulti wifiMulti;
     ...
     wifiMulti.addAP(WIFI_SSID_1, WIFI_PASSWORD_1);   //Connect to WiFi
     WiFi.mode(WIFI_STA);                                                 //Connect to WiFi

It seems GPIO has a conflict with WiFi, as I get the following error:

"
GPIOViewer >> Release 1.5.2
GPIOViewer >> ESP32 Core Version 2.0.16
GPIOViewer >> Chip Model:ESP32-D0WD, revision:1
GPIOViewer >> No PSRAM
GPIOViewer >> Web Application URL is: http://192.168.1.74:8080
Guru Meditation Error: Core  0 panic'ed (Double exception). 

Core  0 register dump:
PC      : 0x4008c226  PS      : 0x00040f36  A0      : 0x8008cc7f  A1      : 0x3ffd40d0  
A2      : 0x3ffb0300  A3      : 0xffffffff  A4      : 0x3ffc52f8  A5      : 0x00060123  
A6      : 0x00060120  A7      : 0x00000001  A8      : 0x40080080  A9      : 0x3ffd41e0  
A10     : 0x00060136  A11     : 0x00040026  A12     : 0x00000001  A13     : 0x0000003c  
A14     : 0x007bf5c0  A15     : 0x003fffff  SAR     : 0x00000004  EXCCAUSE: 0x00000002  
EXCVADDR: 0xffffffe0  LBEG    : 0x4008a0c9  LEND    : 0x4008a0d9  LCOUNT  : 0xfffffffa  

Backtrace: 0x4008c223:0x3ffd40d0 0x4008cc7c:0x3ffd4110 0x4008c223:0x3ffd4130 0x4008c223:0x3ffd4150 0x4008c223:0x3ffd4180 0x4008c223:0x3ffd41a0 0x4008c223:0x3ffd41c0 0x4008c223:0x3ffd41e0 0x4008007d:0x3ffd40d0 0x4009125d:0x3ffd4110 0x4008c223:0x3ffd4130 0x4008c223:0x3ffd4150 0x4008c223:0x3ffd4180 0x4008c223:0x3ffd41a0 0x4008c223:0x3ffd41c0 0x4008c223:0x3ffd41e0 0x4008007d:0x3ffd40d0 0x4009125d:0x3ffd4110 0x4008c223:0x3ffd4130 0x4008c223:0x3ffd4150 0x4008c223:0x3ffd4180 0x4008c223:0x3ffd41a0 0x4008c223:0x3ffd41c0 0x4008c223:0x3ffd41e0 0x4008007d:0x3ffd40d0 0x4009125d:0x3ffd4110 0x4008c223:0x3ffd4130 0x4008c223:0x3ffd4150 0x4008c223:0x3ffd4180 0x4008c223:0x3ffd41a0 0x4008c223:0x3ffd41c0 0x4008c223:0x3ffd41e0 0x4008007d:0x3ffd40d0 0x4009125d:0x3ffd4110 0x4008c223:0x3ffd4130 0x4008c223:0x3ffd4150 0x4008c223:0x3ffd4180 0x4008c223:0x3ffd41a0 0x4008c223:0x3ffd41c0 0x4008c223:0x3ffd41e0 0x4008007d:0x3ffd40d0 0x4009125d:0x3ffd4110 0x4008c223:0x3ffd4130 0x4008c223:0x3ffd4150 0x4008c223:0x3ffd4180 0x4008c223:0x3ffd41a0 0x4008c223:0x3ffd41c0 0x4008c223:0x3ffd41e0 0x4008007d:0x3ffd40d0 0x4009125d:0x3ffd4110 0x4008c223:0x3ffd4130 0x4008c223:0x3ffd4150 0x4008c223:0x3ffd4180 0x4008c223:0x3ffd41a0 0x4008c223:0x3ffd41c0 0x4008c223:0x3ffd41e0 0x4008007d:0x3ffd40d0 0x4009125d:0x3ffd4110 0x4008c223:0x3ffd4130 0x4008c223:0x3ffd4150 0x4008c223:0x3ffd4180 0x4008c223:0x3ffd41a0 0x4008c223:0x3ffd41c0 0x4008c223:0x3ffd41e0 0x4008007d:0x3ffd40d0 0x4009125d:0x3ffd4110 0x4008c223:0x3ffd4130 0x4008c223:0x3ffd4150 0x4008c223:0x3ffd4180 0x4008c223:0x3ffd41a0 0x4008c223:0x3ffd41c0 0x4008c223:0x3ffd41e0 0x4008007d:0x3ffd40d0 0x4009125d:0x3ffd4110 0x4008c223:0x3ffd4130 0x4008c223:0x3ffd4150 0x4008c223:0x3ffd4180 0x4008c223:0x3ffd41a0 0x4008c223:0x3ffd41c0 0x4008c223:0x3ffd41e0 0x4008007d:0x3ffd40d0 0x4009125d:0x3ffd4110 0x4008c223:0x3ffd4130 0x4008c223:0x3ffd4150 0x4008c223:0x3ffd4180 0x4008c223:0x3ffd41a0 0x4008c223:0x3ffd41c0 0x4008c223:0x3ffd41e0 0x4008007d:0x3ffd40d0 0x4009125d:0x3ffd4110 0x4008c223:0x3ffd4130 0x4008c223:0x3ffd4150 0x4008c223:0x3ffd4180 0x4008c223:0x3ffd41a0 0x4008c223:0x3ffd41c0 0x4008c223:0x3ffd41e0 0x4008007d:0x3ffd40d0 0x4009125d:0x3ffd4110 0x4008c223:0x3ffd4130 0x4008c223:0x3ffd4150 0x4008c223:0x3ffd4180 |<-CONTINUES

ELF file SHA256: 08d526a68c8f021b

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13936
load:0x40080400,len:3600
entry 0x400805f0
"

If I remove the lines in my code that have the comment "//Connect to WiFi" the ESP does not reboot/reset;  it would work fine (except I cannot see GPIO Web Page to monitor ESP32 activity).

thelastoutpostworkshop, 2024-05-16:
Try running the example provided with the library to see if the reset persist

jramongit, 2024-05-16:
@thelastoutpostworkshop, do you mean the GPIO_Viewer library?   If so, I already did that with the following lines of code and the issue happens:

#include <gpio_viewer.h>
GPIOViewer gpio_viewer;
...
gpio_viewer.connectToWifi (WIFI_SSID_1, WIFI_PASSWORD_1);   //Connect to WiFi

I also tried including the GPIO_Viewer library, but not using it and trying to connect to WiFi using another library, but issue also happens:

#include <gpio_viewer.h>
GPIOViewer gpio_viewer;
#include <WiFiMulti.h>
WiFiMulti wifiMulti;
 ...
//gpio_viewer.connectToWifi (WIFI_SSID_1, WIFI_PASSWORD_1);   //Connect to WiFi
wifiMulti.addAP(WIFI_SSID_1, WIFI_PASSWORD_1);   //Connect to WiFi
WiFi.mode(WIFI_STA);

If I do not use the GPIO_Viewer library (comment the first 2 lines of code abode), issue is gone, but obviously I cannot monitor my ESP32-WROOM-32 with GPIO_Viewer.

jramongit, 2024-05-16:
@thelastoutpostworkshop, do you mean the GPIO_Viewer library? If so, I already did that with the following lines of code and the issue happens:

```
#include <gpio_viewer.h>
GPIOViewer gpio_viewer;
...
gpio_viewer.connectToWifi (WIFI_SSID_1, WIFI_PASSWORD_1); //Connect to WiFi

```
I also tried including the GPIO_Viewer library, but not using it and trying to connect to WiFi using another library, but issue also happens:

```
#include <gpio_viewer.h>
GPIOViewer gpio_viewer;
#include <WiFiMulti.h>
WiFiMulti wifiMulti;
...
//gpio_viewer.connectToWifi (WIFI_SSID_1, WIFI_PASSWORD_1); //Connect to WiFi
wifiMulti.addAP(WIFI_SSID_1, WIFI_PASSWORD_1); //Connect to WiFi
WiFi.mode(WIFI_STA);
```

If I do not use the GPIO_Viewer library (comment the first 2 lines of code abode), issue is gone, but obviously I cannot monitor my ESP32-WROOM-32 with GPIO_Viewer.

thelastoutpostworkshop, 2024-05-16:
I mean the example provided with the library. 
[image]

[image]

Compile the example et upload it to see if it makes your board reset

thelastoutpostworkshop, 2024-05-16:
It is possible that your code is using pin the way it is not supposed to, some pins have restrictions.  Try adding pin definitions from your code to the example to find it.

jramongit, 2024-05-17:
I ran the example and it wworkd fine.

As per the pins I use are as follows:

D12 - To trigger a Buzzer
D14 - To trigger a Mini Vibrator
D22 - For an RTC module (DS3231)

and the following for a TFT display:
D2
D4
D5
D15
D18
D19
D23

jramongit, 2024-05-17:
I ran the example as you suggested and it worke fine.

As per the pins I use are as follows:

D12 - To trigger a Buzzer
D14 - To trigger a Mini Vibrator
D22 - For an RTC module (DS3231)

and the following for a TFT display:
D2
D4
D5
D15
D18
D19
D23

thelastoutpostworkshop, 2024-05-17:
Post the code here so I can reproduce it on my side.

jramongit, 2024-05-17:
`
#include <gpio_viewer.h>
GPIOViewer gpio_viewer;

#include <WiFi.h>
#include <WiFiMulti.h>
WiFiMulti wifiMulti;

const uint32_t TiempoEsperaWifi = 5000;  // Max time to wait for connecting to WiFi

#include <RTClib.h>  // DS3231 RTC(Real Time Clock)
RTC_DS3231 rtc;  // Create RTC variable

#include <SPI.h>
#include <TFT_eSPI.h>    // Hardware-specific library
//#include "Free_Fonts.h"  // Include the header file attached to this sketch

// Create TFT LCD variable
TFT_eSPI tft = TFT_eSPI();

uint8_t buzzer = 12;    // Pin for Buzzer   (Pin 12, GPIO12)
uint8_t vibrator = 14;  // Pin for Vibrator (Pin 14, GPIO14)

void setup() {

  Serial.begin(115200);  // Enable Serial output for Debug messages

  //------------------------------------------------
  // Conect to WIFI
  //------------------------------------------------
  ///
  /*
  wifiMulti.addAP(WIFI_SSID, WIFI_PASSWORD);
  WiFi.mode(WIFI_STA);
  Serial.print("Conectando a Wifi ..");
  while (wifiMulti.run(TiempoEsperaWifi) != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println(".. Conectado");
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());
  Serial.print(" IP: ");
  Serial.println(WiFi.localIP());
  */

  // Comment the next line if your code aleady includes connection to Wifi in mode WIFI_STA (WIFI_AP and WIFI_AP_STA are not supported)
  gpio_viewer.connectToWifi(WIFI_SSID, WIFI_PASSWORD);

  //------------------------------------------------
  // Initialize the RTC
  //------------------------------------------------
  if(!rtc.begin()) {
      Serial.println("Could not find RTC !  Check circuit.");
      Serial.flush();
      while (1) delay(10);
  }
  rtc.disable32K();  // We don't need the 32K Pin, so disable it
  
  //------------------------------------------------
  // Set pin modes and initialize LEDC
  //------------------------------------------------
  pinMode(buzzer, OUTPUT);    // Set buzzer pin as output
  pinMode(vibrator, OUTPUT);  // Set vibrator pin as output

  // To avoid following error when using tone() function:
  // "E (7244) ledc: ledc_get_duty(740): LEDC is not initialized"
  ledcSetup(0, 10, 8);       //ledcSetup(channel, frequency, bits);
  ledcAttachPin(buzzer, 1);  //ledcAttachPin(pin, channel);

  Serial.println("System start");  // Debug message

  tft.begin();  // Initialize TFT
  tft.fillScreen(0);  //Clear screen
  tft.setRotation(3);  // 0 = No rotation (Portrait),  1 = 90 (left) (Landscape),  2 = 180,  3 = 270 (left) = -90 (right)
  tft.setTextWrap(false);
  tft.setTextColor(0x8E5E);
  tft.setTextDatum(MC_DATUM);  // Configure text to be printed centered
  //tft.setFreeFont(FSS24);  // Select Free Sans 24 point font
  tft.setFreeFont(&FreeSans24pt7b);
  //tft.drawString("Welcome !", 160, 25, GFXFF);
  tft.drawString("Welcome !", 160, 25, 1);

  for (uint8_t i = 0; i < 3; i++) {
    tone(buzzer, 2000, 180);
    delay(180);
    digitalWrite(vibrator, HIGH);
    delay(500);
    digitalWrite(vibrator, LOW);
    delay(250);
  }

  // Must be at the end of your setup
  // gpio_viewer.setSamplingInterval(25); // You can set the sampling interval in ms, if not set default is 100ms
  gpio_viewer.begin();

}  // setup()

//------------------------------------------------------------------------
void loop() {

  delay(5000);

}  // loop()
`

thelastoutpostworkshop, 2024-05-17:
I am able to reproduce the problem, working on it

thelastoutpostworkshop, 2024-05-17:
I dot not have a DS3231, but SDA and SCL is the minimum pins to connect to the ESP32 :
You mentioned it is using D22 (GPIO39), what other pins are you using in your setup ?
[image]
[image]

thelastoutpostworkshop, 2024-05-17:
The problem is now fixed in the latest release (1.5.3).
Somehow the code is setting, indirectly, pin 40 with a channel resolution of 0.
This edge case is now detected by the library avoiding the ESP32 to reset.

jramongit, 2024-05-17:
Just downloaded release 1.5.3, installed and executed again the code.   Issue is resolved !!!
Thanks a lot for your help.

Have a great weekend,

closed 2024-05-17 by jramongit