thelastoutpostworkshop/gpio_viewer #123: Not showing IP address

icaros83, 2024-05-23:
I've included the code as shown and the serial monitor reports "connected to wifi" but doesn't return an IP address.
At first I thought it was because I was using port 8080 elsewhere so I changed it to no avail.
I cannot see the connection on my router so not sure what is going on?
Please see attached screen grab
[image]

Thanks

thelastoutpostworkshop, 2024-05-23:
What is the version of Espressif Arduino are you using for your board ?ç
I see a PSRAM read error, what are the specific configuration for your board in the Arduino IDE ?

icaros83, 2024-05-24:
Thank you for such a quick response.

Do you mean this "esp-2021r2-patch5-8.4.0"?
Not sure what the second question is asking but the board selected is the uPesy ESP32 wrover devkit, I have also tried ESP32 wrover.

icaros83, 2024-05-24:
I've tried the test program and all I get from the ESP32 is a row of dots.

thelastoutpostworkshop, 2024-05-24:
In the Ardnuio IDE, if you go to the boads: smanager:
[image]

You should see the boards version you are using :
[image]

The configuration of your specific board are availaible in the tools menu :
[image]

closed 2024-05-31 by thelastoutpostworkshop