thelastoutpostworkshop/gpio_viewer #124: error: 'analogGetChannel' was not declared

ye9380, 2024-05-30:
I'm getting the following error when I try to compile the example sketch:
`error: 'analogGetChannel' was not declared 
`
my bord is:  ESP32-WROOM-32D

the full error:

```
In file included from 
\Arduino\GPIOViewer\gpioviewer_copy_20240530120619\gpioviewer_copy_20240530120619.ino:9:
\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h: In member function 'int GPIOViewer::readGPIO(int, uint32_t*, pinTypes*)':
c:\Users\ye9380\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:500:33: error: 'analogGetChannel' was not declared in this scope; did you mean 'analogChannel'?
  500 |         uint8_t analogChannel = analogGetChannel(gpioNum);
      |                                 ^~~~~~~~~~~~~~~~
      |                                 analogChannel
```
      
Thanks

thelastoutpostworkshop, 2024-05-30:
You have to use the latest version of the espressif boards 2.0.14 or greater, see this solved issue : https://github.com/thelastoutpostworkshop/gpio_viewer/issues/116#issuecomment-2096921380

ye9380, 2024-05-30:
Im on 3.0.0 ver.
[image]

thelastoutpostworkshop, 2024-05-30:
The library is not compatible yet to 3.0.0

ye9380, 2024-05-30:
It's working now! Thanks!!