thelastoutpostworkshop/gpio_viewer #125: Nothing is shown in serial monitor

vitalickr, 2024-05-30:
Hello,

I compiled and uploaded the example code as is (changed only ssid and pwd) but nothing is showing in the serial monitor window. "Arduino ESP32 Boards" is 2.0.13, "esp32 by Espressif Systems" is 2.0.17 (won't compile with the latest 3.0.0), GPIOViewer is 1.5.3. 

Simple_server example that comes with ESPAsyncWebServer compiles and works just fine with the same credentials.

Sketch uses 796097 bytes (60%) of program storage space. Maximum is 1310720 bytes.
Global variables use 44888 bytes (13%) of dynamic memory, leaving 282792 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port COM4
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b0:a7:32:32:56:d4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000d3fff...
Compressed 18992 bytes to 13110...
Writing at 0x00001000... (100 %)
Wrote 18992 bytes (13110 compressed) at 0x00001000 in 0.3 seconds (effective 567.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.0 seconds (effective 994.6 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 1569.7 kbit/s)...
Hash of data verified.
Compressed 802672 bytes to 508221...
Writing at 0x00010000... (3 %)
Writing at 0x0001b584... (6 %)
Writing at 0x00024c29... (9 %)
Writing at 0x00033097... (12 %)
Writing at 0x000393b8... (15 %)
Writing at 0x0003ef98... (18 %)
Writing at 0x000445fd... (21 %)
Writing at 0x00049bff... (25 %)
Writing at 0x0004ee13... (28 %)
Writing at 0x00054044... (31 %)
Writing at 0x000592e4... (34 %)
Writing at 0x0005e621... (37 %)
Writing at 0x0006392f... (40 %)
Writing at 0x00068dd9... (43 %)
Writing at 0x0006df87... (46 %)
Writing at 0x00073b3e... (50 %)
Writing at 0x000792f0... (53 %)
Writing at 0x0007e57d... (56 %)
Writing at 0x000839b7... (59 %)
Writing at 0x00088d36... (62 %)
Writing at 0x0008e3fb... (65 %)
Writing at 0x00093b1a... (68 %)
Writing at 0x00099479... (71 %)
Writing at 0x0009f1b6... (75 %)
Writing at 0x000a4a74... (78 %)
Writing at 0x000ad3bc... (81 %)
Writing at 0x000b556d... (84 %)
Writing at 0x000ba829... (87 %)
Writing at 0x000c343f... (90 %)
Writing at 0x000c8d8b... (93 %)
Writing at 0x000ce125... (96 %)
Writing at 0x000d3db7... (100 %)
Wrote 802672 bytes (508221 compressed) at 0x00010000 in 6.5 seconds (effective 980.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

thelastoutpostworkshop, 2024-05-30:
Maybe it's a silly question, but what you show above is from the compiler output window, did you open the Serial Monitor in the IDE ?

vitalickr, 2024-05-30:
> Maybe it's a silly question, but what you show above is from the compiler output window, did you open the Serial Monitor in the IDE ?

Of course, there was nothing to show from the serial monitor window as it was empty. I just figured it out: I added a short delay after Serial.begin(115200) in setup() and it started working.

closed 2024-05-31 by thelastoutpostworkshop