thelastoutpostworkshop/gpio_viewer #126: Please add ESP32-C6-Zero Module from https://www.waveshare.com/wiki/ESP32-C6-Zero

HofeBY, 2024-06-08:
Please add ESP32-C6-Zero Module from https://www.waveshare.com/wiki/ESP32-C6-Zero
It would be nice if the back side was also taken into GPIO_Viewer Site, as there are additional GPIO pins to connect to.
Pictures and circuit diagrams are available on the website above.
As a further suggestion, I would be nice to see a clickable link from the original manufacturer on the GPIO_Viewer page.
This allows the user to find additional information directly on the manufacturer's website.
Thank´s for your tool.

Greetings from Germany
HofeBY

thelastoutpostworkshop, 2024-06-08:
The board is now available.
Thanks for the suggestions.

HofeBY, 2024-06-09:
I have to thank you for your wonderful tool.

closed 2024-06-09 by thelastoutpostworkshop