thelastoutpostworkshop/gpio_viewer #129: add esp32 devkit 36 pins

fragolinux, 2024-07-06:
please add this model, very useful with the additional shield... thanks!

[image]

thelastoutpostworkshop, 2024-07-07:
The 36 pins dev board is already there.

fragolinux, 2024-07-11:
> The 36 pins dev board is already there.

ehm... "there" where? i just checked each and every one shown here: https://github.com/thelastoutpostworkshop/gpio_viewer
and none of them is similar to the one i posted (with 3v3, clk and sd0 at the opposite corner of the en one)

thelastoutpostworkshop, 2024-07-11:
Please provide a link to the pinout of this board along with a link showing the board viewed from the top

fragolinux, 2024-07-13:
> Please provide a link to the pinout of this board along with a link showing the board viewed from the top

[image]

from: https://it.aliexpress.com/item/1005006306428914.html

[image]

from: https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

thelastoutpostworkshop, 2024-07-13:
The boars is now available as ESP32 DEVKIT V1 (36 pins)

closed 2024-07-13 by thelastoutpostworkshop

fragolinux, 2024-07-13:
but i don't see it listed, nor new commits to you repo...