thelastoutpostworkshop/gpio_viewer #134: Add Support for Heltec WiFi Kit 32(v3)

CcummingsDT, 2024-08-11:
Hello, and thank you so much for a great debug tool!

I am using the generic view and there are some pins that do not seem to work as expected. The pins on the module work just fine, but are not indicated in the image / web page. I was thinking maybe making the module properly supported would fix that.

The pinout for the device is attached.

I will be sure to buy you a cup of coffee!
[image]

thelastoutpostworkshop, 2024-08-11:
I will add the board.
Which pins should you see on the generic board ?
What kind o pins they are digital, analog, PMW ?

CcummingsDT, 2024-08-11:
Hello!

 

My particular mapping of the module is driven by a very fat sketch with a lot of peripherals attached, so I provide my custom map below that should NOT be used for the general public’s image. All of the pins in green boxes are under sketch control and work fine on this board now, thanks to the confidence your diagnostic tool gave me:

 

 

 

As for the generic view’s issues:

 

Here is a screenshot of the web page and the test sketch I made, since Gpio_viewer does play nice with Heltec’s older 2.0.17 board Support Package I am forced to use (I built this on a fresh PC with all new drivers):

 

 

 

GPIO1 through 5 are Digital Inputs tracking state perfectly on the web page but 6 and 7 are analog inputs yet they show up gray.

 

On the other side of the module, 26, 48 and 47 are set as Digital Outputs but show gray. 33 and 34 are Digital Inputs just like 1-5 but also show up gray. The same for 39 through 46.

 

 

It will still be great to get the Heltec added to your already extensive list of modules.

 

I sent you a small token of my appreciation via the coffee shop.

 

 

 

 

 

 <https://www.dyzinetek.com/> 

Christopher A. Cummings
President
Dyzine Tek, LLC (Funny way to spell Design Tech)
 <http://www.dyzinetek.com> www.dyzinetek.com 

Headquarters
3727 Wildwood Farms DR
Peachtree Corners, GA 30096-6136

Work (833) DYZINE1 (399-4631)
Cell (770) 557-9968
 ***@***.***> ***@***.*** 

Operations Center
2670 North Berkeley Lake RD NW
Suite 18
Duluth, GA 30096

 

This email and any attachments transmitted with it, along with any reply messages associated with this email (collectively, “this email”) are hereby deemed confidential and may be used or stored only by the sender’s intended recipient(s). Any further dissemination to parties not included in the To:, CC or BCC fields of this email or use of this email by persons other than the intended recipient(s) is strictly prohibited. If you have received this email in error, please notify the sender immediately by reply email and then delete this email from your system and destroy any printed copies. Nothing in this transmission is intended to be an electronic signature or to constitute an agreement of any kind under applicable law unless otherwise expressly indicated by an authorized officer of Dyzine Tek LLC. Any views or opinions presented in this email are solely those of the author and do not necessarily represent those of Dyzine Tek LLC. This e-Mail has been scanned for viruses and spyware upon transmission and Dyzine Tek LLC accepts no liability for any damage caused by any Malware received in this email.

thelastoutpostworkshop, 2024-08-11:
Thanks for the coffees!
The screenshots do not show up in your message,
So you are using Arduino Core 2.0.17 ?
Could you provide a small sketch that reproduce the problems on pin 6 and 7 ?

CcummingsDT, 2024-08-11:
Not using the 2.0.17 core with the test sketch. That environment uses 3.03.

 

The DT_gpioviewer_2024-08-10.ino is attached.

 

Not sure why the pics did not come through. Sending as attachments this time.

 

Please let me know if you still do not get them and I will try to zip them up.

 

 

 

 

 

 <https://www.dyzinetek.com/> 

Christopher A. Cummings
President
Dyzine Tek, LLC (Funny way to spell Design Tech)
 <http://www.dyzinetek.com> www.dyzinetek.com 

Headquarters
3727 Wildwood Farms DR
Peachtree Corners, GA 30096-6136

Work (833) DYZINE1 (399-4631)
Cell (770) 557-9968
 ***@***.***> ***@***.*** 

Operations Center
2670 North Berkeley Lake RD NW
Suite 18
Duluth, GA 30096

 

This email and any attachments transmitted with it, along with any reply messages associated with this email (collectively, “this email”) are hereby deemed confidential and may be used or stored only by the sender’s intended recipient(s). Any further dissemination to parties not included in the To:, CC or BCC fields of this email or use of this email by persons other than the intended recipient(s) is strictly prohibited. If you have received this email in error, please notify the sender immediately by reply email and then delete this email from your system and destroy any printed copies. Nothing in this transmission is intended to be an electronic signature or to constitute an agreement of any kind under applicable law unless otherwise expressly indicated by an authorized officer of Dyzine Tek LLC. Any views or opinions presented in this email are solely those of the author and do not necessarily represent those of Dyzine Tek LLC. This e-Mail has been scanned for viruses and spyware upon transmission and Dyzine Tek LLC accepts no liability for any damage caused by any Malware received in this email.

thelastoutpostworkshop, 2024-08-11:
I have added the board, it is called Heltec HT-WB32_V3.
I did not got the zip, you should go to the issue in github to post it, I think the email get rid of the attachements.

CcummingsDT, 2024-08-11:
Wow. I see! It strips everything out, LOL. Let's try this way, instead
[image]
[image]
[image]
[DT_gpioviewer_2024-08-10.zip](https://github.com/user-attachments/files/16576373/DT_gpioviewer_2024-08-10.zip)

thelastoutpostworkshop, 2024-08-11:
Got your code, what module is connected to pin 6 and 7 ?

CcummingsDT, 2024-08-11:
NTC Thermistors

thelastoutpostworkshop, 2024-08-11:
I don't know this module and don't have any.
Do have a link where I could buy this module ?

CcummingsDT, 2024-08-11:
https://heltec.org/project/wifi-kit32-v3/

thelastoutpostworkshop, 2024-08-11:
I don't have your Haltec board, but when I run your code, I got values from pin 6 and 7 (pins are floating):
[image]

thelastoutpostworkshop, 2024-08-11:
I tested it on an ESP32-S3

CcummingsDT, 2024-08-11:
Cool! I will retry in my morning!
I am on the East Coast of the USA, In Georgia, so I get in the office before 9 AM tomorrow.

CcummingsDT, 2024-08-11:
What you have set up here is really very impressive. Do you freelance dev at all?

thelastoutpostworkshop, 2024-08-11:
Thanks, no I don't do freelance dev, too busy!

CcummingsDT, 2024-08-11:
Going offline for the evening. Going to watch Wheel of Time on Amazon Prime.

CcummingsDT, 2024-08-12:
Good afternoon! I loaded up some supper simple Output exerciser code to toggle the output lines and everything there seems to be working great. I do seem to have to set the outputs HIGH before the gray dot is replaced with a red one, and then subsequent drops to LOW are properly recorded.

Also, still not love on the Analog Inputs 6 and 7.

https://www.dropbox.com/scl/fi/s4cihemgkaais2kd3rugv/20240812_143558.mp4?rlkey=m39sq54b2si9wy38lgwzk3c45&dl=0

thelastoutpostworkshop, 2024-08-12:
Can you show me the output of the GPIOViewer you get on the Serial Monitor ?

CcummingsDT, 2024-08-12:
https://github.com/user-attachments/assets/c080c2fc-f436-489e-8a0b-32851b8fce91

thelastoutpostworkshop, 2024-08-12:
Can you give show me on the serial monitor the data sent to pin 6 and 7 ?

CcummingsDT, 2024-08-12:
Not sure if the video is broken on your end, but sure does not play on mine. Here is a Link:

[image]

CcummingsDT, 2024-08-12:
That link was not correct, LOL,

As for the values coming from the pins 6 and 7, were you able to watch the first video? Those numbers scrolling by on the serial monitor are what I was getting back.

[image]

CcummingsDT, 2024-08-12:
[image]

CcummingsDT, 2024-08-12:
[image]

CcummingsDT, 2024-08-12:
My context menu video dropbox link process in Windows is broken.

https://www.dropbox.com/scl/fi/j7m8hppjiy50f6y8b6ch9/20240812_171649.mp4?rlkey=j5fh9orvoftq9cf2yvlnrlz7k&st=don1jasm&dl=0

thelastoutpostworkshop, 2024-08-12:
I have a good idea of where the problem is, I will do some testing on my side.

CcummingsDT, 2024-08-12:
OK. Thanks Very Much!!

thelastoutpostworkshop, 2024-08-13:
I have updated the code, you can test v1.0.6 by downloading the library and install it (v1.0.6 is still not officially release since I need to do more test).

CcummingsDT, 2024-08-13:
OK. I will be sure to give it a try and report the results I get. Thanks again!

thelastoutpostworkshop, 2024-08-14:
> 20240813_085346.jpg (view on web)
> 20240813_085351.jpg (view on web)
>
> Works like a charm! Thanks, Charles!
>
> I am wondering, for true PWM and lines for I2C / SPI, do you show
> something different? Obviously the animation would not work well for very
> fast toggling signals

Glad it works.
It already work for PWM.  For I2C and SPI I did not explore it yet.

closed 2024-08-15 by thelastoutpostworkshop