thelastoutpostworkshop/gpio_viewer #135: XIAO_ESP32C3

w0gen, 2024-08-24:
I am getting the following error:
board selection is XIAO_ESP32C3
It is a seeed studio unit
.GPIOViewer >> Connected to WiFi
GPIOViewer >> Release 1.5.6
GPIOViewer >> ESP32 Core Version 3.0.4
GPIOViewer >> Chip Model:ESP32-C3, revision:4
GPIOViewer >> No PSRAM
GPIOViewer >> Web Application URL is: http://192.168.4.198:8080
E (1256) adc_oneshot: adc_oneshot_new_unit(87): adc unit not supported

thelastoutpostworkshop, 2024-08-24:
Does the web application work ?
Can you confirm that this board does not have ADC function ?

w0gen, 2024-08-24:
Sorry I am a newbe...
this is my exact code that i am running:
/***
This example is intended to demonstrate the use of the GPIO Viewer Library.

Tutorial : https://youtu.be/UxkOosaNohU
Latest Features : https://youtu.be/JJzRXcQrl3I
Documentation : https://github.com/thelastoutpostworkshop/gpio_viewer
***/

#include <gpio_viewer.h> // Must me the first include in your project
#include <ESPAsyncWebServer.h>
#include <WiFi.h>
#include <AsyncTCP.h>

GPIOViewer gpio_viewer;

void setup()
{
  Serial.begin(115200);

  // Comment the next line, If your code aleady include connection to Wifi in mode WIFI_STA (WIFI_AP and WIFI_AP_STA are not supported)
  gpio_viewer.connectToWifi("xxxxx", "xxx!xxx");
  // gpio_viewer.setPort(5555);                 // You can set the http port, if not set default port is 8080

  // Your own setup code start here

  // Must be at the end of your setup
  // gpio_viewer.setSamplingInterval(25); // You can set the sampling interval in ms, if not set default is 100ms
  gpio_viewer.begin();
}

// You don't need to change your loop function
void loop() {

}
// The rest of your code here

thelastoutpostworkshop, 2024-08-24:
but does the web application works, if you copy the URL given in the serial monitor into your web browser ?

w0gen, 2024-08-24:
web application is not working...

w0gen, 2024-08-24:
Yes web app is working, i didn't add 8080

thelastoutpostworkshop, 2024-08-24:
I have a XIAO board, I will give it a try tonight

thelastoutpostworkshop, 2024-08-24:
in the meantime, add this define before including the header of the GPIOViewer library, like this and try again:
#define NO_PIN_FUNCTIONS
#include <gpio_viewer.h> // Must me the first include in your project

w0gen, 2024-08-24:
added no pin functions per your last comment:
getting these errors:
[ 36123][E][esp32-hal-periman.c:180] perimanGetPinBus(): Invalid pin: 24
[ 36129][E][esp32-hal-periman.c:180] perimanGetPinBus(): Invalid pin: 24
[ 36137][E][esp32-hal-gpio.c:190] __digitalRead(): IO 24 is not set as GPIO.
[ 36144][E][esp32-hal-periman.c:180] perimanGetPinBus(): Invalid pin: 25
[ 36151][E][esp32-hal-periman.c:180] perimanGetPinBus(): Invalid pin: 25
[ 36157][E][esp32-hal-gpio.c:190] __digitalRead(): IO 25 is not set as GPIO.
[ 36165][E][esp32-hal-periman.c:180] perimanGetPinBus(): Invalid pin: 26
[ 36171][E][esp32-hal-periman.c:180] perimanGetPinBus(): Invalid pin: 26
[ 36178][E][esp32-hal-gpio.c:190] __digitalRead(): IO 26 is not set as GPIO.

w0gen, 2024-08-24:
[130017][E][esp32-hal-adc.c:218] __analogInit(): adc_oneshot_new_unit failed with error: 258
[130031][E][esp32-hal-adc.c:275] __analogRead(): Analog initialization failed!
[130038][E][esp32-hal-gpio.c:190] __digitalRead(): IO 5 is not set as GPIO.
[130045][E][esp32-hal-gpio.c:190] __digitalRead(): IO 6 is not set as GPIO.
[130053][E][esp32-hal-gpio.c:190] __digitalRead(): IO 7 is not set as GPIO.
[130060][E][esp32-hal-gpio.c:190] __digitalRead(): IO 8 is not set as GPIO.
[130067][E][esp32-hal-gpio.c:190] __digitalRead(): IO 9 is not set as GPIO.
[130074][E][esp32-hal-gpio.c:190] __digitalRead(): IO 10 is not set as GPIO.
[130081][E][esp32-hal-gpio.c:190] __digitalRead(): IO 11 is not set as GPIO.
[130088][E][esp32-hal-gpio.c:190] __digitalRead(): IO 12 is not set as GPIO.
[130095][E][esp32-hal-gpio.c:190] __digitalRead(): IO 13 is not set as GPIO.
[130102][E][esp32-hal-gpio.c:190] __digitalRead(): IO 14 is not set as GPIO.
[130109][E][esp32-hal-gpio.c:190] __digitalRead(): IO 15 is not set as GPIO.

w0gen, 2024-08-24:
[image]

thelastoutpostworkshop, 2024-08-24:
after adding the define the web application worked ?

w0gen, 2024-08-24:
I didn't add :8080 to the ip address...
the define makes no difference
Here is what my monitor shows...
The web is working
nnecting to WiFi...
.GPIOViewer >> Connected to WiFi
GPIOViewer >> Release 1.5.6
GPIOViewer >> ESP32 Core Version 3.0.4
GPIOViewer >> Chip Model:ESP32-C3, revision:4
GPIOViewer >> No PSRAM
GPIOViewer >> Web Application URL is: http://192.168.4.198:8080
E (1262) adc_oneshot: adc_oneshot_new_unit(87): adc unit not supported
E (1364) adc_oneshot: adc_oneshot_new_unit(87): adc unit not supported

thelastoutpostworkshop, 2024-08-24:
ok so your good to go using it, ignore the error messages in the serial monitor for now. 
I will fix it and let you know when the fix is ready so you can test it.

w0gen, 2024-08-24:
Thank you Sir!

thelastoutpostworkshop, 2024-08-24:
After reviewing this issue, there is no way I can remove these error messages.
Since the ESP32 Core does not provide suitable APIs for all type of pins, the GPIOViewer library try to read gpio pins to see if it can  detect values, raising errors on some pins.
This does not affect the final result in the web application.
You can also remove these messages by compiling the code with "Core Debug LEvel" to "None" : 
[image]

closed 2024-08-26 by thelastoutpostworkshop