thelastoutpostworkshop/gpio_viewer #138: Conflict with OneWire?

sarelvdwalt, 2024-09-04:
I suspect there is some sort of a conflict with the OneWire library. When I comment out my `gpio_viewer.begin();` line (i.e. everything is loaded and set up, but I don't start the viewer) I get the normal 22 degrees or whatever the DS18B20 measures. But, the moment I have that line in the code I get -127. From looking at the OneWire code, that means "Device not connected".

Here is my code for sample:
```
#include <gpio_viewer.h>
#include <OneWire.h>
#include <DallasTemperature.h>

GPIOViewer gpio_viewer;

// GPIO where the DS18B20 is connected to
const int oneWireBus = 33;

// Setup a oneWire instance to communicate with any OneWire devices
OneWire oneWire(oneWireBus);

// Pass our oneWire reference to Dallas Temperature sensor 
DallasTemperature sensors(&oneWire);

void setup() {
  // Start the Serial Monitor
  Serial.begin(115200);

  // Start the DS18B20 sensor
  sensors.begin();

  // Connect to WiFi:
  gpio_viewer.connectToWifi("REDACTED", "REDACTED");
  gpio_viewer.begin();
}

void loop() {
  sensors.requestTemperatures(); 
  float temperatureC = sensors.getTempCByIndex(0);
  float temperatureF = sensors.getTempFByIndex(0);
  Serial.println(temperatureC);
  delay(1000);
}
```

Sample output of "normal" operation with it commented out:
```
21.75
21.75
21.75
21.75
21.75
```

Sample output of "broken" operation when it is enabled and running:
```
GPIOViewer >> Connected to WiFi
GPIOViewer >> Release 1.5.6
GPIOViewer >> ESP32 Core Version 3.0.4
GPIOViewer >> Chip Model:ESP32-D0WDQ6, revision:100
GPIOViewer >> No PSRAM
GPIOViewer >> Web Application URL is: http://192.168.107.132:8080
-127.00
GPIOViewer >> Connected, sampling interval is 100ms
-127.00
-127.00
-127.00
-127.00
```

thelastoutpostworkshop, 2024-09-04:
Yes there is probable a conflict, a DS18B20, but I will try to find a way to test it.
Can you try with version 1.5 of the GPIOViewer library to see if you get the same results ?

sarelvdwalt, 2024-09-05:
Hey hey, ok so I had success. The following versions had a compiler issue:

- 1.5.0
- 1.5.1
- 1.5.2
- 1.5.3

The error was:
```
In file included from /Users/myname/Documents/Arduino/sarel/sarel.ino:1:
/Users/myname/Documents/Arduino/libraries/GPIOViewer/src/gpio_viewer.h: In member function 'int GPIOViewer::readGPIO(int, uint32_t*, pinTypes*)':
/Users/myname/Documents/Arduino/libraries/GPIOViewer/src/gpio_viewer.h:500:33: error: 'analogGetChannel' was not declared in this scope; did you mean 'analogChannel'?
  500 |         uint8_t analogChannel = analogGetChannel(gpioNum);
      |                                 ^~~~~~~~~~~~~~~~
      |                                 analogChannel

exit status 1

Compilation error: exit status 1
```

When I tried `1.5.4` it compiled, and upon running it I got a good result:
```
GPIOViewer >> Connected to WiFi
GPIOViewer >> Release 1.5.4
GPIOViewer >> ESP32 Core Version 3.0.4
GPIOViewer >> Chip Model:ESP32-D0WDQ6, revision:100
GPIOViewer >> No PSRAM
GPIOViewer >> Web Application URL is: http://192.168.107.132:8080
19.00
19.06
19.00
19.06
19.06
```

Tried version `1.5.5` and that was fine too:
```
GPIOViewer >> Connected to WiFi
GPIOViewer >> Release 1.5.5
GPIOViewer >> ESP32 Core Version 3.0.4
GPIOViewer >> Chip Model:ESP32-D0WDQ6, revision:100
GPIOViewer >> No PSRAM
GPIOViewer >> Web Application URL is: http://192.168.107.132:8080
19.06
19.06
19.06
```

So it appears the bug is with `1.5.6` specifically.

thelastoutpostworkshop, 2024-09-05:
ok thanks for doing the test,  Do you see pin activity on your GPIO 33 (oneWireBus) in the GPIOViewer web application ?

sarelvdwalt, 2024-09-05:
Any time :) Yes, I do see activity on it - but only after a while. 

- v1.5.5 (where it's working fine) - I see activity immediately, and it's binary (so it shows "High" with a red circle)
- v1.5.6 (where it's broken) - after initial startup it doesn't show anything, and after a while (around a minute later at most) it shows Analog red, value 4095.

thelastoutpostworkshop, 2024-09-05:
Thank again it will help.  I do not know yet when I will be able to test it, stay tuned!

smitzkar, 2025-01-05:
I have run into pretty much the exact same issue using GPIOViewer@^1.5.8 on XiaoESP32-S3-Sense. 

I used platformIO at first, then checked Arduino IDE 2.3.4 and noticed that the **port was flashing connected/disconnected like mad.** 

Using the Generic View, I saw activity on Pins 1-20 which seemed to just pick up random noise and especially touch. Using the board specific view gave similar results for 1-9 (no other shown). 

I commented out all of my own code and the issue was persistent with nothing but GPIOViewer: 

```C++
#include <gpio_viewer.h> 
GPIOViewer gpio_viewer;

#include <Arduino.h>

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println("Testing serial print");

  gpio_viewer.connectToWifi("my-ssid", "hunter2");

  //...

  gpio_viewer.begin();
}

void loop() {
// ...
}
```

thelastoutpostworkshop, 2025-01-05:
Try by adding this define before including the library
#define NO_PIN_FUNCTIONS
#include <gpio_viewer.h> // Must me the first include in your project

This will disable the detection of pin functions, this maybe what is causing the problem.

smitzkar, 2025-01-05:
Thanks for the amazingly quick response!

I tried using 1.5.5 first and it worked, then also tried your advice ( #define NO_PIN_FUNCTIONS ) and this seems to fix the issue for 1.5.8! 

---  
Just to maybe provide some extra info:

For both 1.5.5 and 1.5.8 (with your fix), GPIO0**9** (one of the available pins on xiaos3) is shown as slowly switching high/low (non-constant rhythm, roughly 1-5s). When changing to generic view, after letting it run for a few minutes, it also showed 29 and 31 as high on 1.5.5, 29 high, 30 low on 1.5.8. 
The viewer also shows reaction to touch on **9**-13 (high on touching any of them, overriding the high/low of 9, no output to serial monitor). 

Finally, showing up only in Arduino IDE serial monitor (using the working 1.5.5), I got the following: 
```
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037a8f6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1188
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbf0
load:0x403cb700,len:0x30e4
entry 0x403c88ac
Testing serial print
GPIOViewer >> Connecting to WiFi...
..GPIOViewer >> Connected to WiFi
GPIOViewer >> Release 1.5.5
GPIOViewer >> ESP32 Core Version 3.1.0
GPIOViewer >> Chip Model:ESP32-S3, revision:2
GPIOViewer >> No PSRAM

assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1851 (Required to lock TCPIP core functionality!)

Backtrace: 0x403764ce:0x3fca9a80 0x4037cc2d:0x3fca9aa0 0x40383645:0x3fca9ac0 0x42023857:0x3fca9bf0 0x420239c1:0x3fca9c10 0x4200cf64:0x3fca9c30 0x4200c38d:0x3fca9c80 0x42004cb1:0x3fca9ca0 0x42004dfa:0x3fca9d20 0x42010f43:0x3fca9d40 0x4037d91a:0x3fca9d60

ELF file SHA256: a059c5f35

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
```

repeating endlessly, but with a working web viewer!

 ---

My issue is fixed, so thank you very much, but I'm available, if you need any further info!

thelastoutpostworkshop, 2025-01-05:
Thanks for reporting the issue.