thelastoutpostworkshop/gpio_viewer #142: md5 compiler error

StepanHlava, 2024-12-14:
c:\Users\Hlava22\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp: In function 'bool getMD5(uint8_t*, uint16_t, char*)':
c:\Users\Hlava22\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:74:3: error: 'mbedtls_md5_starts_ret' was not declared in this scope; did you mean 'mbedtls_md5_starts'?
   74 |   mbedtls_md5_starts_ret(&_ctx);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_starts
c:\Users\Hlava22\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:75:3: error: 'mbedtls_md5_update_ret' was not declared in this scope; did you mean 'mbedtls_md5_update'?
   75 |   mbedtls_md5_update_ret(&_ctx, data, len);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_update
c:\Users\Hlava22\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:76:3: error: 'mbedtls_md5_finish_ret' was not declared in this scope; did you mean 'mbedtls_md5_finish'?
   76 |   mbedtls_md5_finish_ret(&_ctx, _buf);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_finish

StepanHlava, 2024-12-14:
may be solved in WebAuthentication.cpp
  // mbedtls_md5_starts_ret(&_ctx);
  // mbedtls_md5_update_ret(&_ctx, data, len);
  // mbedtls_md5_finish_ret(&_ctx, _buf);
  mbedtls_md5_starts(&_ctx);
mbedtls_md5_update(&_ctx, data, len);
mbedtls_md5_finish(&_ctx, _buf);

thelastoutpostworkshop, 2024-12-14:
Thanks for the tip

closed 2024-12-24 by thelastoutpostworkshop