thelastoutpostworkshop/gpio_viewer #148: Serial2 stops working after gpioViewer.begin();

homeplayer2, 2024-12-26:
before gpioViewer.begin(); in setup() 

serial2.begin(115200, SERIAL_8N1, 16, -1, true);
will setup and read the serial buffer.

after starting gpioViewer.begin(); communication on serial2 will stop

thelastoutpostworkshop, 2024-12-26:
What board are you using and what is the version in the board manager ?

homeplayer2, 2024-12-26:
ESP32 Systeeminformatie
Systeem
Uptime: 0 dagen, 0 uur, 49 min, 0 sec.
Chip Model: ESP32-D0WD-V3
Chip Revisie: 301
CPU Frequentie: 80 MHz
Core Versie: 3.1.0
SDK Versie: 5.3.1.241024
Flash Geheugen
Flash Grootte: 4 MB (Totaal flash geheugen)
Flash Speed: 40 MHz
Flash Mode: 2
Geheugen
Vrij Heap Geheugen: 217200 bytes (Dynamisch geheugen beschikbaar voor het programma)
Minimaal Vrij Heap Geheugen (sinds boot): 192572 bytes (Laagste hoeveelheid vrij heap geheugen ooit)
Heap Gebruik: 37% (Percentage Heap in gebruik)
Sketch Grootte: 702048 bytes (Grootte van de geÃ¼ploade code)
Vrije Sketch Ruimte: 1310720 bytes (Ruimte over in het flash geheugen voor OTA updates)
WiFi
WiFi SSID: --Snip--
WiFi Kanaal: 6
WiFi IP Adres: 192.168.2.2
WiFi Netmasker: 255.255.255.0
WiFi Gateway: 192.168.2.254
RSSI (WiFi Signaalsterkte): -64 dBm
MAC Adres: --snip--
Hostname: p1meter
WiFi Status: 3
WiFi Modus: 1
WiFi BSSID: --snip--
WiFi Autoconnect: 1

platformIo.ini
[env]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
lib_deps = 
        SD
        SPI
        xairooo/SimpleTime updated @ ^1.1.0
	https://github.com/mathieucarbou/AsyncTCP.git
        thelastoutpostworkshop/GPIOViewer@^1.5.8
build_flags = 
 

[env:OTA]
upload_protocol = espota
upload_port = p1meter

[env:USB]
upload_speed = 921600

--snip--@pop-os:~/Documents/PlatformIO/Projects/24-12$ pio pkg list
Resolving OTA dependencies...
Platform espressif32 @ 2024.11.31 (required: espressif32)
├── framework-arduinoespressif32 @ 3.1.0+sha.1c74f314 (required: https://github.com/tasmota/arduino-esp32/releases/download/3.1.0.241117/framework-arduinoespressif32.zip)
├── framework-espidf @ 3.50301.0 (required: https://github.com/tasmota/esp-idf/releases/download/v5.3.1.241024/esp-idf-v5.3.1.zip)
├── tool-cmake @ 3.21.3 (required: platformio/tool-cmake @ ~3.21.0)
├── tool-esptoolpy @ 4.8.1 (required: https://github.com/tasmota/esptool/releases/download/v4.8.1.1/esptool.zip)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 3.2.0 (required: tasmota/tool-mklittlefs @ ^3.2.0)
├── tool-ninja @ 1.7.1 (required: platformio/tool-ninja @ ^1.7.0)
├── tool-openocd-esp32 @ 2.1100.20220706 (required: platformio/tool-openocd-esp32 @ ~2.1100.0)
├── tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 (required: platformio/tool-riscv32-esp-elf-gdb @ 14.2.0+20240403)
├── tool-scons @ 4.40801.0 (required: platformio/tool-scons @ ~4.40801.0)
├── tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 (required: platformio/tool-xtensa-esp-elf-gdb @ 14.2.0+20240403)
├── toolchain-esp32ulp @ 1.23800.240113 (required: platformio/toolchain-esp32ulp @ ~1.23800.0)
└── toolchain-xtensa-esp-elf @ 13.2.0+20240530 (required: platformio/toolchain-xtensa-esp-elf @ 13.2.0+20240530)

Libraries
├── AsyncTCP @ 3.3.1+sha.22a6988 (required: git+https://github.com/mathieucarbou/AsyncTCP.git)
├── GPIOViewer @ 1.5.8 (required: thelastoutpostworkshop/GPIOViewer @ ^1.5.8)
│   ├── Async TCP @ 3.1.4 (required: Async TCP)
│   └── ESP Async WebServer @ 1.2.4 (required: ESP Async WebServer)
│   │   └── AsyncTCP @ 1.1.1 (required: me-no-dev/AsyncTCP @ ^1.1.1)
└── SimpleTime updated @ 1.1.0 (required: xairooo/SimpleTime updated @ ^1.1.0)

Resolving USB dependencies...
Platform espressif32 @ 2024.11.31 (required: espressif32)
├── framework-arduinoespressif32 @ 3.1.0+sha.1c74f314 (required: https://github.com/tasmota/arduino-esp32/releases/download/3.1.0.241117/framework-arduinoespressif32.zip)
├── framework-espidf @ 3.50301.0 (required: https://github.com/tasmota/esp-idf/releases/download/v5.3.1.241024/esp-idf-v5.3.1.zip)
├── tool-cmake @ 3.21.3 (required: platformio/tool-cmake @ ~3.21.0)
├── tool-esptoolpy @ 4.8.1 (required: https://github.com/tasmota/esptool/releases/download/v4.8.1.1/esptool.zip)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 3.2.0 (required: tasmota/tool-mklittlefs @ ^3.2.0)
├── tool-ninja @ 1.7.1 (required: platformio/tool-ninja @ ^1.7.0)
├── tool-openocd-esp32 @ 2.1100.20220706 (required: platformio/tool-openocd-esp32 @ ~2.1100.0)
├── tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 (required: platformio/tool-riscv32-esp-elf-gdb @ 14.2.0+20240403)
├── tool-scons @ 4.40801.0 (required: platformio/tool-scons @ ~4.40801.0)
├── tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 (required: platformio/tool-xtensa-esp-elf-gdb @ 14.2.0+20240403)
├── toolchain-esp32ulp @ 1.23800.240113 (required: platformio/toolchain-esp32ulp @ ~1.23800.0)
└── toolchain-xtensa-esp-elf @ 13.2.0+20240530 (required: platformio/toolchain-xtensa-esp-elf @ 13.2.0+20240530)

Libraries
├── AsyncTCP @ 3.3.1+sha.22a6988 (required: git+https://github.com/mathieucarbou/AsyncTCP.git)
├── GPIOViewer @ 1.5.8 (required: thelastoutpostworkshop/GPIOViewer @ ^1.5.8)
│   ├── Async TCP @ 3.1.4 (required: Async TCP)
│   └── ESP Async WebServer @ 1.2.4 (required: ESP Async WebServer)
│   │   └── AsyncTCP @ 1.1.1 (required: me-no-dev/AsyncTCP @ ^1.1.1)
└── SimpleTime updated @ 1.1.0 (required: xairooo/SimpleTime updated @ ^1.1.0)

Is did what you asked of me?

thelastoutpostworkshop, 2024-12-26:
I am unable to reproduce it, what is connected on Serial2 and why only RX pin (16) is defined in your example ?

homeplayer2, 2024-12-27:
Dutch smart Meter (DSMR) energy, 
and i also trying to set up some extra inputs and outputs,

Thanks for your hard work, I'll be testing my esp today, just gpio viewer and OTA 
To be honest i am a Frankenstein coder, i can read code, i cant write it.

here some screen shots.

[image]
[image]
[image]

homeplayer2, 2024-12-27:
whit the following code;
`
#include <gpio_viewer.h>
#include <Arduino.h>
#include <ArduinoOTA.h>
#include <WiFi.h>

#include "../secrets.h"
#define HOSTNAME "p1meter"

GPIOViewer gpioViewer;

void setup() {
  WiFi.setHostname(HOSTNAME);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

  while (WiFi.status() != WL_CONNECTED) {
    delay(50);
  }
  pinMode(15, INPUT_PULLUP); // a test PIN will be low in production
  digitalWrite(15, HIGH);    // set to high
  ArduinoOTA.begin();
  ArduinoOTA.onStart([]() { });
  gpioViewer.begin();
}

void loop() {
  ArduinoOTA.handle();
  delay(2);
}

`
GPIO-pin 15 will show as Inactive pin, and gray, its only after i disconnect gnd and connect gnd to it that it will show its true color, 
[image]
not sure about the pins top left what thats about

thelastoutpostworkshop, 2024-12-27:
Thanks for the information. since pin 15 is set HIGH before calling gpioViewer.begin() and if there is nothing happening on pin 15 after that, nothing will show on the image of the board.  When you disconnect pin 15 and reconnect it, it is captured and shown, but if nothing happens in the code with this pin, nothing will be shown.   The pins on the left show floating values (it happens when pins are not used)

closed 2025-01-01 by thelastoutpostworkshop