thelastoutpostworkshop/gpio_viewer #149: ESP32 Device stuck in reboot loop

anant-j, 2025-01-01:
Hi,
After compiling, my ESP32 device gets stuck in a constant boot loop.
Device Specifications: 
- Board: Adafruit Huzzah32 Feather v2 
- Libraries used: 
  - AsyncTCP @ v3.3.1
  - GPIOViewer @ v1.5.8
  - ESPAsyncWebServer @ v3.4.5
  - Arduino IDE @ v2.3.4
  - Board Manager: esp32 @ 3.1.0
Here is the serial log:
```
03:45:08.846 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
03:45:08.846 -> configsip: 271414342, SPIWP:0xee
03:45:08.846 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
03:45:08.881 -> mode:DIO, clock div:1
03:45:08.881 -> load:0x3fff0030,len:4688
03:45:08.881 -> load:0x40078000,len:15460
03:45:08.881 -> ho 0 tail 12 room 4
03:45:08.881 -> load:0x40080400,len:4
03:45:08.881 -> load:0x40080404,len:3196
03:45:08.881 -> entry 0x400805a4
03:45:09.570 -> Start Initializing...
03:45:09.650 -> GPIOViewer >> Connecting to WiFi...
03:45:10.148 -> .GPIOViewer >> Connected to WiFi
03:45:10.398 -> GPIOViewer >> Release 1.5.8
03:45:10.398 -> GPIOViewer >> ESP32 Core Version 3.1.0
03:45:10.398 -> GPIOViewer >> Chip Model:ESP32-PICO-V3-02, revision:300
03:45:10.398 -> GPIOViewer >> PSRAM Size 2.00 MB
03:45:10.398 -> GPIOViewer >> Web Application URL is: http://192.168.58.111:8080
03:45:10.430 -> Initialization complete...
03:45:10.711 -> 
03:45:10.711 -> assert failed: xEventGroupClearBits event_groups.c:495 (xEventGroup)
03:45:10.711 -> 
03:45:10.711 -> 
03:45:10.711 -> Backtrace: 0x40082709:0x3ffcf690 0x4009625d:0x3ffcf6b0 0x4009c746:0x3ffcf6d0 0x40097311:0x3ffcf800 0x400f46f6:0x3ffcf820 0x400f47b6:0x3ffcf860 0x400e87a9:0x3ffcf880 0x400e865f:0x3ffcf930 0x400d31cd:0x3ffcf950 0x400d38e8:0x3ffcf970 0x400f5ac4:0x3ffcf990 0x4009747a:0x3ffcf9b0
03:45:10.748 -> 
03:45:10.748 -> 
03:45:10.748 -> 
03:45:10.748 -> 
03:45:10.748 -> ELF file SHA256: 35c15bb0a
03:45:10.748 -> 
03:45:10.946 -> Rebooting...
```

Any help with this issue is much appreciated.
Thanks in advance.

thelastoutpostworkshop, 2025-01-01:
Paste a minimal sketch that reproduce the crashing, so i can test it

anant-j, 2025-01-02:
Hi, I noticed the minimal example sketch did not cause the same issues. This seems to be conflicting with some other libraries I use (I believe https://github.com/T-vK/ESP32-BLE-Keyboard). 
Closing the issue as I could not figure out a way to make these two work together with my logic in setup.

closed 2025-01-02 by anant-j